adafruit/circuitpython #1644: Varying duty_cycle on PWMOut very occasionally results in a high blip

kevinjwalters, 2019-03-13:
I've been using PWMOut for audio use and I heard a few anomalies on `4.0.0-beta.3`. I've looked into it briefly and it appears that very occasionally there's a high blip on my PWM. @deshipu asked if it's one cycle on Discord and I think it is although it's difficult to tell on my traditional (non storage) oscilloscope. It can be recreated with this code on a Circuit Playground Express (CPX):

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.3 on 2019-03-07; Adafruit CircuitPlayground Express with samd21g18
>>> import board, pulseio, time
>>> p6 = pulseio.PWMOut(board.A6, duty_cycle=32768, frequency=100, variable_frequency=True)
>>> while True:
...     for i in range(4096, 28672, 1000):
...         p6.duty_cycle = i
...         time.sleep(0.1)
...     for i in range(28672, 4096, -1000):
...         p6.duty_cycle = i
...         time.sleep(0.1)
````

If you watch the square wave for about 30 seconds you're likely to see one or two. I'm going to check another pin out too...

kevinjwalters, 2019-03-13:
I can connect up a logic analyser and try and capture it to check duration of the high if this is needed?

kevinjwalters, 2019-03-13:
From discord: 

> @sommersoft   well on the CPX, A6 (PB09) is a TC only, where A1 (PA05) is a TCC only. so, that could be a vector to dive into

dhalbert, 2019-03-14:
@kevinjwalters I'll check the TC code, but could you also check using pin `A1` and see if you also see a blip? I'm not sure the TC's have the same kind of double-buffering synchronization you need to avoid glitches like this.

kevinjwalters, 2019-03-15:
I've checked A1 and didn't noticed any problems over a 4 minute period.

kevinjwalters, 2019-03-16:
I put an Ikalogic SQ25 logic analyser on A6 to verify what I was hearing and seeing on non-storage oscilloscope, it does appear to be a single cycle blip which has the effect of joing two either side into one long high pulse.

[image]

dhalbert, 2019-03-17:
The TC's are not double-buffered like the TCC's, so changing the duty cycle can cause these glitches. There's some discussion of a workaround here: https://medium.com/@nebsp/smoothly-changing-a-timers-frequency-on-the-arduino-zero-1f6cd285369a, though I have to study this a bit.

kevinjwalters, 2019-03-17:
Thanks for looking at this. I'm still surprised these are so noticeable when the pwm is used for sound, I only found this because I heard the audio being a bit clicky.

tannewt, 2019-03-27:
@dhalbert Do you plan on doing this for 4.0? Seems like we could do this in 4.x.

dhalbert, 2019-03-27:
I'll move it to 4.x. I know what I want to do, but I need to test it too.

dhalbert, 2019-04-05:
@kevinjwalters I am experimenting with adjusting the TC count when the duty cycle is modified, to make smoother transitions. Would you like to try this UF2? I haven't tested it myself.
[cpx-smooth-pwmout.uf2.zip](https://github.com/adafruit/circuitpython/files/3049189/cpx-smooth-pwmout.uf2.zip)
It's a build of https://github.com/dhalbert/circuitpython/tree/smooth-tc-pwmout

kevinjwalters, 2019-04-05:
Sure, might have time this weekend to have a look, thanks.

kevinjwalters, 2019-04-15:
I've put that uf2 on one CPX but not got the board connected to the relevant hardware at the moment. I'll get back to this in a few days.

kevinjwalters, 2019-06-11:
Took me a while to get back to this but just found my electronics and reassembled with CPX and worked out I had a duff breadboard track. 

Just tested your version and it's far, far worse!

dhalbert, 2019-06-21:
In what way? Could you attach a test program and describe a setup so I can listen for the issues? That would save me from trying changes without testing.

jepler, 2019-10-24:
I assume this is what is happening: 
[image]

ladyada, 2019-10-24:
yah sometimes the chip can have a double-buffer so top/count are only updated on wraparound. the other option is to write to CC0 when you update top/count, they you may end up with a short blip. i dont think busy-waiting till you wrap-around is a great idea, but its an idea.