adafruit/circuitpython #1650: CircuitPython 4.0 Beta 4 hard crash

iraytrace, 2019-03-17:
When running on a CircuitPlayground Express with:

> Adafruit CircuitPython 4.0.0-beta.4 on 2019-03-15; Adafruit CircuitPlayground Express with samd21g18

Got a hard crash which said:

> You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Host system was a Mac.  Code was being loaded from MU to the board when the error message reported.

main.py is attached
[main_py.txt](https://github.com/adafruit/circuitpython/files/2974741/main_py.txt)

makermelissa, 2019-03-17:
Thanks @iraytrace. I believe this will be fixed with PR #1649. My suggestion is to use Beta 3 for the moment as I was having far fewer problems with that one.

dhalbert, 2019-03-17:
beta.5 to fix this is now out.