adafruit/circuitpython #1651: microcontroller.nvm Broken in Alphas/Beta 4.0

kdb424, 2019-03-17:
I have been trying to use this function, and have tested from alpha 2 to beta 4 (the latest at time of issue) and have not been able to successfully use this. Depending on the version, different issues happen from REPL lockups, to rebooting to safe mode. Is this not supported at the current time?

```python
import microcontroller
microcontroller.nvm[0:3] = b"\xcc\x10\x00"
print(microcontroller.nvm[0]
```

dhalbert, 2019-03-18:
On which board(s) are you having trouble?

kdb424, 2019-03-18:
Whoops, forgot to say. This is on the Itsy Bitsy M4 express so far that I have tested.

tannewt, 2019-03-20:
@kdb424 Thanks for the report. It should still be working. I plan on fixing it today.

tannewt, 2019-03-20:
@kdb424 A quick test with Beta 5 on a different M4 worked just fine. Is there more to your code than the three lines you posted?

tannewt, 2019-03-20:
I haven't been able to reproduce this on an Itsy Bitsy M4 either. @kdb424 please reopen/comment if you can still trigger it with Beta 5 or newer. Thanks!

closed 2019-03-20 by tannewt