adafruit/circuitpython #1652: 4.0 beta 4 hangs and code modified

gvcp, 2019-03-17:
Several issues with Feather M4 express (also simple blink aplication) on MacOS (10.12.6):
When using the mu editor it frequently hangs when trying to save code.py.
Sometimes code can be saved but is frequently modified to all 0xFF´s with exactly the same length than the  original code. 
Serial communication with Mu hangs, ctrl C crl D has no effect in this state.
Uodated Mu from 1.0 to 1.02, but doesn´t improve anything.
When using some other editor with the CIRCUITPTY volume code is correctly downloaded and executed directly after a reset of the Feather, but when saving it a second time the same problem appears. (volume hangs, code modified to FF´s.

Had  similar behavior with beta 3 but seems that behavior has become worse

jerryneedell, 2019-03-17:
This may be related to #1643

dhalbert, 2019-03-17:
@gvcp Please try again with the beta.5 release (new this morning), and let us know here if that fixed it.

gvcp, 2019-03-17:
Doesn't appear any more with blink application, seems to behave stable in this respect like CP3.x. 
Just wondering about the connection with BLE stated in  #1643 as I don´t have any BLE hardware connected to Feather M4 express.

dhalbert, 2019-03-17:
Thanks. The BLE issues in #1643 are probably different, it appears.

closed 2019-03-17 by dhalbert