adafruit/circuitpython #1655: GitHub credentials for Adabot now wrong?

dhalbert, 2019-03-17:
The first Travis subjob is failing trying to make a release PR for circuitpython.org:
https://travis-ci.com/adafruit/circuitpython/jobs/185484652. I heard second-hand a credential got changed?
```
$ (! var_search "${TRAVIS_TESTS-}" website || (cd tools && python3 build_board_info.py && cd ..))
Not logged in
{"message":"Bad credentials","documentation_url":"https://developer.github.com/v3"}
Traceback (most recent call last):
  File "build_board_info.py", line 265, in <module>
    generate_download_info()
  File "build_board_info.py", line 206, in generate_download_info
    git_info, current_info = get_current_info()
  File "build_board_info.py", line 88, in get_current_info
    raise RuntimeError("cannot get master sha")
RuntimeError: cannot get master sha
The command "(! var_search "${TRAVIS_TESTS-}" website || (cd tools && python3 build_board_info.py && cd ..))" exited with 1.
```

tannewt, 2019-03-18:
Ya, I removed one set of adabot creds I showed on my stream. Will update them tomorrow. Thanks for handling it!

tannewt, 2019-03-19:
I updated the creds for adafruit/circuitpython and adafruit/adabot so they should be good to go.

closed 2019-03-19 by tannewt