adafruit/circuitpython #1656: atmel-samd: Cannot disable pulseio or rotaryio (link error)

jrcutler, 2019-03-17:
It appears to be impossible to build an `atmel-samd` target with either `CIRCUITPY_PULSEIO = 0` or `CIRCUITPY_ROTARYIO = 0` due to hard-coded references to functions in `samd-peripherals/samd/external_interrupts.c` (causing link errors).

To replicate, add `CIRCUITPY_PULSEIO = 0` or `CIRCUITPY_ROTARYIO = 0` to `ports/atmel-samd/boards/trinket_m0`, then try to build for `trinket_m0`.

I was able to build using the following modification to `external_interrupt_handler` to elide references to the relevant functions.
```c
void external_interrupt_handler(uint8_t channel) {
    uint8_t handler = channel_handler[channel];
    if (handler == EIC_HANDLER_PULSEIN) {
#if CIRCUITPY_PULSEIO
        pulsein_interrupt_handler(channel);
#endif
    } else if (handler == EIC_HANDLER_INCREMENTAL_ENCODER) {
#if CIRCUITPY_ROTARYIO
        incrementalencoder_interrupt_handler(channel);
#endif
    }
    EIC->INTFLAG.reg = (1 << channel) << EIC_INTFLAG_EXTINT_Pos;
}
```

Using this modification, I found the following space savings (for a `trinket_m0`-based target):
- `CIRCUITPY_PULSEIO = 0`: 6200 bytes
- `CIRCUITPY_ROTARYIO = 0`: 208 bytes

_Note_: I **do not** recommend this modification as-is. There is almost certainly a better way to handle this, such as building trivial placeholders for these functions when the components are disabled (or using weak symbols). Ideally, `samd-peripherals` should be further decoupled from the `common-hal` directory in CircuitPython (or merged back into CircuitPython).

dhalbert, 2019-03-18:
Thanks for this. Filed an issue in the other repo as well: https://github.com/adafruit/samd-peripherals/issues/19

tannewt, 2019-08-23:
This was fixed by #1745

closed 2019-08-23 by tannewt