adafruit/circuitpython #1657: Fatal exception and reset when 'import ntptime'.

water5, 2019-03-17:
When import 'ntptime' package, appear message below and board reset:
-------------------------------
Fatal exception 3(LoadStoreErrorCause):
...
-------------------------------

test board:  ESP8266 module

circuitpython version:  adafruit-circuitpython-feather_huzzah-3.1.2.bin

dhalbert, 2019-03-18:
There seems to be something wrong with the frozen-in version of `ntptime`. As a workaround for now, try copying the file below,`ntptime.py`, to the root directory and importing it. I did that and it didn't crash. This is the exactly the file that is supposed to be built in:

https://github.com/adafruit/circuitpython/blob/3.x/ports/esp8266/modules/ntptime.py

Link to raw version (for direct download):
https://github.com/adafruit/circuitpython/raw/3.x/ports/esp8266/modules/ntptime.py

dhalbert, 2019-03-18:
It appears that all of the frozen python modules files are broken, so if you need others, copy those too, from the `modules` directory above.

dhalbert, 2019-03-18:
This is broken all the way back to 3.0.0. Alternatively, try the 2.3.1 release for esp8266, which was well-tested: https://github.com/adafruit/circuitpython/releases/tag/2.3.1

ESP8266 support was dropped for CircuitPython 4.0.

jerryneedell, 2019-03-18:
this is strange because I know I have used many of those modules in the past and I think I was using them well beyond 3.0.0

dhalbert, 2019-03-18:
I was wrong, only `ntpsetup.py` and `port_diag.py` cause a crash. The other frozen modules are fine.

water5, 2019-03-19:
Actually, ntptime.settime() also can't work, it dependent utime module, Adafruit circuitpython has n't utime module.

dhalbert, 2019-03-19:
@water5 - Aha, you've figured out the problem! Try just replacing `utime` with `time`.

I'm not sure what the problem is with `port_diag.py`, but actually calling `esp.check_fw()` calls a crash. However, I don't think that should cause the import to crash.

water5, 2019-03-19:
`import ntptime
import time as utime

ntptime.settime()`

also appear :
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ntptime.py", line 32, in settime
ImportError: no module named 'utime'

dhalbert, 2019-03-19:
You have to use your own copy of ntptime.py, and change the `import utime` that's further down in the program. You can't use the built-in frozen copy.

water5, 2019-03-19:
When `time.localtime()` appear "RuntimeError: RTC is not supported on this board".
But actually, `machine.RTC().datetime()` works fine.

ladyada, 2019-08-23:
hiya, looks like you figured it out & we no longer support the esp8266 as of 4.x - please check out micropython for esp8266 support! thank you :)

closed 2019-08-23 by ladyada