adafruit/circuitpython #1658: No rule to make target 'common-hal/analogio/__init__.c

rrace001, 2019-03-18:
Just made a new clone the repository and got an error.  I can still make boards on an version a few months older.

make: *** No rule to make target 'common-hal/analogio/__init__.c', needed by 'build-trinket_m0/genhdr/qstr.i.last'.  Stop.

dhalbert, 2019-03-18:
I think your clone is incomplete somehow. Does the file `circuitpython/ports/atmel-samd/common-hal/analogio/__init__.c` exist? It's part of the source tree. Is this is a clone from `adafruit/circuitpython` or your own repo?

rrace001, 2019-03-18:
This was a fork from your repo and the files existed. Since you thought the clone was incomplete then I decided to delete all and redo the clone to make sure. It works fine now.  Not sure what I did wrong but thanks for the quick reply.

closed 2019-03-18 by rrace001

dhalbert, 2019-03-18:
Glad it’s ok!