adafruit/circuitpython #1660: board error? "ValueError: TFT_BACKLIGHT in use"

Rathmore, 2019-03-18:
My trying to get my Hallowing to run the slideshow code found here: https://learn.adafruit.com/getting-started-with-hallowing-for-hackaday-supercon-attendees/slideshows

```
import board
from adafruit_slideshow import PlayBackOrder, SlideShow
import pulseio

# Create the slideshow object that plays through once alphabetically.
slideshow = SlideShow(board.DISPLAY, pulseio.PWMOut(board.TFT_BACKLIGHT),
    folder="/images", loop=True,
    order=PlayBackOrder.ALPHABETICAL, dwell=2.5)

while slideshow.update():
    pass

```

I'm running the beta 5 build of CP4 on my Hallowing with the latest CP4 libraries.

When I run the code, I get:
```
Traceback (most recent call last):
  File "code.py", line 7, in <module>
ValueError: TFT_BACKLIGHT in use 
```

(The REPL is displaying on the screen with a small CP logo at the top (which is new?) and the backlight is operating. I'm doing my coding in Mu 1.0.2.)

Thanks, and let me know if you need any other info.

-Dan

makermelissa, 2019-03-18:
Updating the code to what I have below gets past the `ValueError: TFT_BACKLIGHT in use` error and gives `RuntimeError: All timers in use` which I believe is on the pulseio.PWMOut() call.
```
import board
from adafruit_slideshow import PlayBackOrder, SlideShow
import pulseio
import displayio

displayio.release_displays()

# Create the slideshow object that plays through once alphabetically.
slideshow = SlideShow(board.DISPLAY, pulseio.PWMOut(board.TFT_BACKLIGHT),
					  folder="/images", loop=True,
					  order=PlayBackOrder.ALPHABETICAL, dwell=2.5)

while slideshow.update():
    pass
```

Full Error Message:
```
Traceback (most recent call last):
  File "code.py", line 9, in <module>
RuntimeError: All timers in use
```

tannewt, 2019-03-18:
Hi hi, this "BACKLIGHT in use" error is due to us moving control of the backlight into the display object itself (so it can adjust the backlight when user code isn't running).

I believe the latest slideshow library has been updated for this and you should just remove the `pulseio.PWMOut(board.TFT_BACKLIGHT),` parameter from code.py.

tannewt, 2019-03-19:
@Rathmore did my response above get you going on this? I don't think it's a bug.

Rathmore, 2019-03-19:
I removed the parameter and now get:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 9, in <module>
  File "adafruit_slideshow.py", line 200, in __init__
  File "adafruit_slideshow.py", line 316, in advance
TypeError: extra keyword arguments given
```
I updated my libraries to the version that dropped today (3/19), and I still get the same output. I am still on beta 5.

My slideshow line reads 
```
slideshow = SlideShow(board.DISPLAY, folder="/images", order=PlayBackOrder.ALPHABETICAL, loop=True, dwell=2.5)
```
Thanks.

tannewt, 2019-03-20:
@Rathmore Progress! Looks like @jerryneedell has already fixed the lib and it just needed a release. So, grab 1.0.4 [here](https://github.com/adafruit/Adafruit_CircuitPython_Slideshow/releases/tag/1.0.4) and give it a try. It'll be in the bundle sometime tomorrow.

Closing this because it's a library issue and has been fixed.

closed 2019-03-20 by tannewt