adafruit/circuitpython #1663: add gmtime() to time module

CedarGroveStudios, 2019-03-18:
Add `gmtime()` to support foundational timekeeping in UTC to permit conversion to `localtime()` or when needed to calculate an offset to convert a UTC time value to the local timezone. See discussion: [adafruit/Adafruit_CircuitPython_PyPortal/issues #16](https://github.com/adafruit/Adafruit_CircuitPython_PyPortal/issues/16)

jerryneedell, 2019-10-11:
I was looking into this and want to make sure I understand what is needed.
As far as I can tell, the time module only knows about "local time" it has no concept of Time Zone.

However -- when we have an internet connection and us the adafruit_ntp library to set the time, it sets it to GMT. 

Do we need to add Time zone awareness and then return localtime or gmtime as requested?

I am thinking that was is now referred to as localtime should really be gmtime and if local time requested, use the Time Zone setting to offset it.. 

Is that what you have in mind? Or have I missed the point entirely...

CedarGroveStudios, 2019-10-12:
I think you've got it. The primary objective is to be able to determine local time accurately based on gmtime when it's available. The nice-to-have would be to calculate gmtime from local time when an actively-updated gmtime isn't available. 

Since the flag is available, calculating and including the DST offset would be frosting on the cake.
`>>> time.localtime()
Out[2]: time.struct_time(tm_year=2019, tm_mon=3, tm_mday=18, tm_hour=0, tm_min=3, tm_sec=26, tm_wday=0, tm_yday=77, tm_isdst=1)
`

calcut, 2022-02-15:
Slightly confused - are we saying that currently `time.localtime(posix_seconds)` always returns in UTC?
i.e. does it do the same as `time.gmtime(posix_seconds)`?

if not, how does it figure out what local time should be? it doesn't seem to automagically pick up the timezone from secrets.py

(my timezone is currently == UTC, which makes it harder to spot potential issues!)