adafruit/circuitpython #1668: adafruit_hid.mouse doesn't run when board is plugged into USB

uhrheber, 2019-03-19:
I tried the following code:
```
from time import sleep
from adafruit_hid.mouse import Mouse
 
mouse = Mouse()

while True:
    mouse.move(x=15)
    sleep(0.07)
    mouse.move(y=15)
    sleep(0.07)
    mouse.move(x=-15)
    sleep(0.07)
    mouse.move(y=-15)
    sleep(0.07)
```

Just a simple test, to move the mouse cursor around in squares.
The code runs perfectly when:
- started by ^D
- started by file system interrupt when the file containing the code (here: main.py) is copied to the virtual drive, or an existing main.py is edited and saved (e.g. in Mu).

But: When the board is freshly connected to USB, the code doesn't start.
Tested on ItsyBisty M4 Express and pca10059.
After plugging in, the M4 Express shows a green LED, followed by an orange LED, followed by four blue flashes.

HID keyboard does work, though.

deshipu, 2019-03-19:
I don't suppose you have a PyPortal or Hallowing to try and get the messages displayed on the screen? I tried with a Hallowing just now, but I couldn't reproduce your problem — it works for me.

deshipu, 2019-03-19:
By the way, which version of the firmware are you using?

uhrheber, 2019-03-19:
Latest master from today.

deshipu, 2019-03-19:
That's the same I tried. I suppose this is something specific to Windows.

deshipu, 2019-03-19:
You could try adding some delay at the beginning of the program...

uhrheber, 2019-03-19:
Maybe. I'll try it with Linux when I'm home.

uhrheber, 2019-03-19:
I already tried a 5 seconds delay at the beginning. No dice.

uhrheber, 2019-03-19:
I now added a 0.5 seconds delay before and after instantiating mouse, and now it works.

This should maybe be added to the documentation/examples.

The working code is now:
```
from time import sleep
 
import analogio
import board
import digitalio
from adafruit_hid.mouse import Mouse

sleep(0.5)
mouse = Mouse()
sleep(0.5) 
 
while True:
    mouse.move(x=15)
    sleep(0.07)
    mouse.move(y=15)
    sleep(0.07)
    mouse.move(x=-15)
    sleep(0.07)
    mouse.move(y=-15)
    sleep(0.07)
 ```       

I'm closing this issue.

closed 2019-03-19 by uhrheber

dhalbert, 2019-03-19:
The Mouse code actually does this already, but it sounds like you need a bit more: https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/master/adafruit_hid/mouse.py#L60