adafruit/circuitpython #1669: displayio: infinite loop in Bitmap.c

dhalbert, 2019-03-19:
`shared-bindings/displayio/Bitmap.c`:
```
STATIC mp_obj_t displayio_bitmap_make_new(const mp_obj_type_t *type, size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args) {
    mp_arg_check_num(n_args, kw_args, 3, 3, false);
    uint32_t width = mp_obj_get_int(pos_args[0]);
    uint32_t height = mp_obj_get_int(pos_args[1]);
    uint32_t value_count = mp_obj_get_int(pos_args[2]);
    uint32_t power_of_two = 1;
    while (value_count > (1U << power_of_two)) {
        power_of_two <<= 1;
    }
```
I don't think  `1U << power_of_two` is meant here, since it's shifting by a very large amount, maybe just `power_of_two`? It rapidly shifts too far and will become 0, and then `power_of_two` will also be come 0 as the 1 gets shifted too far.

@deshipu was trying `image = displayio.Bitmap(8, 16, 65546)`.

deshipu, 2019-03-19:
I think it's supposed to be ``+=1``. The idea is to find the lowest power of two that is greater than the ``value_count``.

tannewt, 2019-03-20:
Please remember to add a milestone.

dhalbert, 2019-03-20:
Sorry about that. We could also add milestones to some PR's, though I'm not sure about the status of a number of them.

tannewt, 2019-03-20:
np, the main issue is that I won't see it if it's only assigned to me. When I'm looking at what to do next I start from the milestone issue list.

deshipu, 2019-03-20:
By the way, how about renaming the ``power_of_two`` to ``bits`` or even ``bpp`` — this way there won't be a confusion whether the number says which power of two this is, or the value of that power of two.

commit 81fe8060 (2019-03-26): Properly calculate BPP for displayio.Bitmap

Fix #1669
closed 2019-03-26 by tannewt