adafruit/circuitpython #1671: displayio.Palette not checking color_count

caternuson, 2019-03-20:
Can instantiate a `Palette` with a `color_count` that exceeds max (255?) without getting any errors. But then it errors out later.
```python
Adafruit CircuitPython 4.0.0-beta.5 on 2019-03-17; Adafruit PyPortal with samd51j20
>>> from displayio import Palette
>>> p1 = Palette(255)
>>> p2 = Palette(256)
>>> p1[0] = 0xFF0000
>>> p2[0] = 0xFF0000
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
IndexError: Palette index out of range
>>> 
```

deshipu, 2019-03-20:
This is strange, because I don't see anything that would limit the size of the palette — ``uint32_t`` is used as the index type everywhere.

deshipu, 2019-03-20:
Ah, I see, ``color_count`` is ``uint8_t`` in the ``displayio_palette_t`` struct.

PR #1676 (2019-03-20): Make displayio.Palette support more than 255 colors
Fix #1671
commit f5cfd6e1 (2019-03-20): Make displayio.Palette support more than 255 colors

Fix #1671
closed 2019-03-20 by tannewt