adafruit/circuitpython #1679: PyPortal crash -- USB disconnects

jerryneedell, 2019-03-22:
With current Master 
```
Adafruit CircuitPython 4.0.0-beta.5-17-gfce63b17c on 2019-03-21; Adafruit PyPortal with samd51j20
```
I have been running the openweather demo on a pyportal. After running for a few hours, it will suddenly drop the USB connection and the program appears to have stopped (not updates on screen. CIRCUITPY is no linger mounted and /dev/ttyACM0 is no longer accessible

It recovers and runs normally after a hard RESET (press RESET button)

dmesg report at the time of failure - same time as last screen update.
```
[Fri Mar 22 04:37:50 2019] usb 3-3.2.3: reset full-speed USB device number 19 using xhci_hcd
[Fri Mar 22 04:38:06 2019] usb 3-3.2.3: device descriptor read/64, error -110
[Fri Mar 22 04:38:21 2019] usb 3-3.2.3: device descriptor read/64, error -110
[Fri Mar 22 04:38:22 2019] usb 3-3.2.3: reset full-speed USB device number 19 using xhci_hcd
[Fri Mar 22 04:38:37 2019] usb 3-3.2.3: device descriptor read/64, error -110
[Fri Mar 22 04:38:53 2019] usb 3-3.2.3: device descriptor read/64, error -110
[Fri Mar 22 04:38:53 2019] usb 3-3.2.3: reset full-speed USB device number 19 using xhci_hcd
[Fri Mar 22 04:38:58 2019] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Mar 22 04:39:04 2019] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Mar 22 04:39:04 2019] usb 3-3.2.3: device not accepting address 19, error -62
[Fri Mar 22 04:39:04 2019] usb 3-3.2.3: reset full-speed USB device number 19 using xhci_hcd
[Fri Mar 22 04:39:09 2019] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Mar 22 04:39:15 2019] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[Fri Mar 22 04:39:15 2019] usb 3-3.2.3: device not accepting address 19, error -62
[Fri Mar 22 04:39:15 2019] usb 3-3.2.3: USB disconnect, device number 19
[Fri Mar 22 04:39:15 2019] cdc_acm 3-3.2.3:1.0: ttyACM0: USB ACM device
[Fri Mar 22 04:39:15 2019] usb 3-3.2.3: new full-speed USB device number 26 using xhci_hcd

```

tannewt, 2019-03-27:
Is this pixel_shader related too?

trimmkm, 2019-04-09:
I've seen similar behaviour under 4.0.0 Beta 6, running both the quotes code and weather station code.  After several hours of running, the code would hang, and the NeoPixel status light would be steady yellow.  A hard reset would fix the issue.  When running from USB power supply only, it may run for 24 hours between crashes.

trimmkm, 2019-04-15:
Just to follow up on my earlier post, I think my issue is related to my WiFi access point.  I tried to replicate my issues on a second pyportal, and was having difficulty accessing the password protected WiFi.  Both pyportals ran fine on a guest (unsecured) WiFi network.  My original pyportal, which would crash faithfully within 24 hours of starting, has been running for several days now without issue.

tannewt, 2019-04-17:
Thanks for the follow up @trimmkm.

I'm going to close this because it is hard to reproduce. If someone comes up with a quicker way to reproduce please post a new issue.

closed 2019-04-17 by tannewt