adafruit/circuitpython #1680: PyPoertal crash -- EvenCountdown example

jerryneedell, 2019-03-22:
I am finding the the PyPortal EventCoundown example causes a (HardFault) reboot to safe mode wit the current CP Master build.  https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/PyPortal_EventCountdown  Warning -- it may also corrupt the FS.  I have had varying results with this -- Sometimes it runs but the background splash disappears. Sometimes it crashes....

Run on:
Adafruit CircuitPython 4.0.0-beta.5-17-gfce63b17c on 2019-03-21; Adafruit PyPortal with samd51j20

dhalbert, 2019-03-24:
I have tried this a few times. I am consistently seeing the background disappear, but otherwise it hasn't crashed. Does it crash right away, or should I let it run a while?

jerryneedell, 2019-03-24:
@dhalbert  With the default settings, it does not crash immediately -- infact, it seems to be running OK today,
However, if I cahange the event date to:
```
14,16c14,16
< EVENT_MONTH = 6
< EVENT_DAY = 28
< EVENT_HOUR = 17
---
> EVENT_MONTH = 4
> EVENT_DAY = 15
> EVENT_HOUR = 9
```
it crashes immediately on first update. That is, I see the background, then after a few seconds, it crashes.

jerryneedell, 2019-03-24:
very strange -- I  run eventcountdown.py on the PyPortal, it runs OK -- I then copied eventcountdown.py to new_copy.py  -- When I run new_copy.py  it crashes...
a diff of the files does not report any difference.

In all cases, I am running this from the REPL not code.py.

even stranger,  any of the examples (that is different dates) run OK if loaded as code.py.....
My head hurts...

deshipu, 2019-03-24:
I think it might depend on where the file is in the partition table, how it is split into blocks, and other things like that. On a non-express board you could just copy the ``current.uf2`` file to reproduce the whole filesystem exactly, perhaps we could have some tool to dump the filesystem off the flash chip on the express boards as well?

dhalbert, 2019-03-25:
The essence of this bug is described in #1695. I don't know about the hard crashes, but that may be yet some other bug, or else it's a manifestation of the #1695 bug. Could close this in favor of #1680.

dhalbert, 2019-03-27:
I think this is fixed by #1714. Could you reopen if you are still seeing crashes? Thanks.

closed 2019-03-27 by dhalbert