adafruit/circuitpython #1681: serial_write_substring hangs when writing to disconnected USB

dhalbert, 2019-03-22:
If CDC USB is connected and then becomes disconnected, `serial_write_substring()` in `supervisor/shared/serial.c` will loop forever trying to write a string. The call to `tud_cdc_connected()` returns `true` even though USB has been disconnected.

Opening an issue in tinyusb about this.

#1680 discusses a case where PyPortal hangs (no display update, but no crash) after running the OpenWeather demo for a long time. In discussing this with @jerryneedell, the host computer went to sleep during that period, with the REPL connected. I replicated the problem this way:

1. Start PyPortal running OpenWeather. Use a USB cable with a data/power-only switch
2. Connect to REPL.
3. After a minute or less, disconnect the data by flipping the switch on the USB cable. The PyPortal will remain powered but will lose its USB connection.
4. The PyPortal will eventually try to `print()` a string, and will call `serial_write_substring()`. As described above, it will loop forever there, because `tud_cdc_connected()` but no bytes are able to be written to CDC.

jerryneedell, 2019-03-22:
@dhalbert If you run from a "code.py" and never connect to USB will this problem be avoided. That is, do you have to open a REPL session in order to trigger this?

dhalbert, 2019-03-23:
Yes, I believe it has to do with being connected and then disconnecting.

jerryneedell, 2019-03-24:
Just FYI, I have been running the slideshowsd.py example as code.py on a PyPortal with current master with no data connection, just a wall supply. After a few hours, the slideshow stops and I see on the screen that it has rebooted to safe mode. The reason is not displayed. If I then do a hard reset, it restarts the code.py slideshow.
No idea if this is related, but I wanted to report it since this does not involve a USB connection.

hathach, 2019-03-27:
@dhalbert I just find out the dcd_samd21 doesn't report disconnection/unplug event to the stack. So yes, if you provide the power via DC jack to Metro M0 then unplug the usb cable. The stack will still think it is connected. Hmm, very tricky is I couldn't find an way to detect such event reading through samd21 family data sheet. The closest thing I could detect is suspend interrupt, which will always happen as part of disconnection, but it can be a standalone bus event (e.g windows when to standby, it will send SUSPEND event, and send RESUME after wakeup without the need to re-enumeration).

dhalbert, 2019-03-27:
I was thinking we could detect that a write (and maybe a read) was stalled, and set a flag that says it's disconnected.

Is this only samd21 or also samd51? Is this inherent in the peripheral or is it an ASF4 issue?

hathach, 2019-03-27:
@dhalbert STALL is bit signal, same as ACK or NAK. Physical disconnect will cause the bus in SE0 ( D+ and D- in LOW). Somehow It also generate the suspend event (J state for 3 ms) with samd. I couldn't find way to detect usb disconnect with samd21 and samd51, but I am not familiar with them. Maybe you or other may know better.

dhalbert, 2019-03-27:
I don't mean a literal USB STALL, just a write that isn't making progress (zero characters written).

hathach, 2019-03-27:
imo, suspend is much better for this. I am looking more closer now. Maybe it is somewhere else, power module or something..

dhalbert, 2019-04-05:
Fixed by #1748.

closed 2019-04-05 by dhalbert