adafruit/circuitpython #1683: Displayio column & row commands are 16 bit (need 8 bit too)

rdagger, 2019-03-24:
I wrote a CircuitPython displayio driver for the SSD1351 OLED 128x128 display.  I confirmed the init sequence works by manually sending SPI commands to the display using the _FourWire.send()_ method.  However, none of the displayio commands work.

A closer look with a scope shows that displayio sends 16 bit values when it calls _set_column_ and _set_row_.  The SSD1351 expects 8 bit value for column and row.

Would it be possible to add a command_byte_length parameter to the _displayio.display_ class constructor?  It could default to 2 for 16 bit displays such as the ILI9341 and the user could pass 1 for 8 bit displays like the SSD1351.
[image]

makermelissa, 2019-03-24:
@rdagger I noticed there's a comment in CircuitPython with this issue with a "To Do" next to it, so let's leave this open. However, I was able to get a working driver without this change for the SSD1351. As of right now the [PR](https://github.com/adafruit/Adafruit_CircuitPython_SSD1351/pull/3) hasn't been merged yet, but you can download it off my repo until it has.

rdagger, 2019-03-25:
@makermelissa I tested the SSD1351 driver and the example that you linked above and it doesn't work yet.  I don't see how it could until the changes to _set_column_ & _set_row_ byte length are added?  

Maybe I'm misunderstanding your post or maybe you have a different revision of the SSD1351?  I'm using a generic SSD1351, but the Adafruit SSD1351 datasheet also shows a single byte for column/row start and a single byte for column/row end (page 36, section 10.1.1 & 10.1.2).

dhalbert, 2019-03-25:
Could be 4.0.0 milestone or could be 4.x, depending on whether this appears to be necessary.

makermelissa, 2019-03-25:
Thanks for testing @rdagger. It must have been the CircuitPython PR I prepared for this. I'll test it with the PR again.

makermelissa, 2019-03-25:
Oh, I didn't address your comment @dhalbert. I'd say it should remain a milestone since it's necessary for displayio to work on certain displays.

@rdagger I swear it was working last night, but it's not working for me anymore, so now I'm trying to figure out what I did so I can duplicate. :)

rdagger, 2019-03-25:
Seems like it will be difficult for the displayio.display class to maintain compatibility across all the different LCD displays.  I took a look at writing a driver for an older Adafruit ST7565 display.  Instead of a _set_column_ command it has _set_column_upper_ and _set_column_lower_ commands.  Instead of _set_row_ it has _set_page_ (The display is broken into 8 horizontal pages which are numbered from the center 0 to 3 and 7 to 4).

deshipu, 2019-03-25:
Fortunately there is a very limited number of different displays in use in there, and all the new ones follow the MIPI standard, so we could either hard-code the weird ones or simply completely leave them out.

makermelissa, 2019-03-25:
In certain cases such as this display, I think there is enough flexibility to get it working. I had it displaying the other night, but must not have unplugged the display before testing Beta 5, so something was still properly initialized when I tested. I do know it works under the right conditions though.

makermelissa, 2019-03-26:
Ok, I figured it out. I was using && instead of & for bitwise and. Once I fixed that it worked beautifully. Submitted PR so hopefully we'll get that up and running pretty soon.

PR #1708 (2019-03-29): Added option to allow Displays with Single Byte Boundaries
This fixes #1683 and allows the SSD1351 display to work with displayio.
closed 2019-03-29 by tannewt