adafruit/circuitpython #1686: PyPortal -- safe mode during slideshow

jerryneedell, 2019-03-24:
I have been running the slideshowsd.py example bleow  as code.py on a PyPortal with current master with no data connection, just a wall supply. After a few hours, the slideshow stops and I see on the screen that it has entered safe mode. The reason is not displayed. If I then do a hard reset, it restarts the code.py slideshow normally.
May be related to #1681  but no USB connection is necessary and once started, this code does not write to the REPL.

CP Build:
```
Adafruit CircuitPython 4.0.0-beta.5-17-gfce63b17c on 2019-03-21; Adafruit PyPortal with samd51j20
```

this is the code running:
```
import os
import board
import busio
import digitalio
import storage
import adafruit_sdcard
from adafruit_slideshow import PlayBackOrder, SlideShow

""" If you want to use an SD card to store the images: """
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

IMAGE_DIRECTORY = "/sd/images"
#IMAGE_DIRECTORY = "/images"

def print_directory(path, tabs=0):
    for file in os.listdir(path):
        stats = os.stat(path + "/" + file)
        filesize = stats[6]
        isdir = stats[0] & 0x4000

        if filesize < 1000:
            sizestr = str(filesize) + " by"
        elif filesize < 1000000:
            sizestr = "%0.1f KB" % (filesize / 1000)
        else:
            sizestr = "%0.1f MB" % (filesize / 1000000)

        prettyprintname = ""
        for _ in range(tabs):
            prettyprintname += "   "
        prettyprintname += file
        if isdir:
            prettyprintname += "/"
        print('{0:<20} Size: {1:>6}'.format(prettyprintname, sizestr))

        # recursively print directory contents
        if isdir:
            print_directory(path + "/" + file, tabs + 1)

print_directory(IMAGE_DIRECTORY)

# Create the slideshow object that plays through once alphabetically.
slideshow = SlideShow(board.DISPLAY, None,
            folder=IMAGE_DIRECTORY, loop=True,
            order=PlayBackOrder.RANDOM, dwell=2.5)

while slideshow.update():
    pass
```

deshipu, 2019-03-24:
I had similar problems while developing the snake game for the Hallowing M0 Express. The code is at https://github.com/deshipu/hallowing-snake

dhalbert, 2019-03-25:
I ran the example above overnight. It didn't go into safe mode but it did throw an `IOError` 22 exception on this line in `adafruit_slideshow.py`:
```
            self._image_file = open(image_name, "rb")
```
I suspect something got smashed. I'm rerunning with a breakpoint on the lowest-level exception handling so I can catch something like this.

dhalbert, 2019-03-26:
@jerryneedell How many images were in your slideshow?

jerryneedell, 2019-03-26:
Four

dhalbert, 2019-03-26:
Thanks, I was seeing the exception above within half an hour or so with 13 images; reduced it to 12, and it's running for a couple of hours, but doesn't seem critical. In the bug above, the value passed in instead of the `image_name` string was the integer `14`, which is suspiciously close to 13. Hence my try to reduce it by one and see if I see something similar.

dhalbert, 2019-03-29:
I let this run overnight, unconnected to USB, and it didn't crash. There are still problems related to USB connect/disconnect (#1681), but #1721 or other fixes may have fixed the underlying problem here.

tannewt, 2019-04-04:
Anyone still worried about this or can I close it?

jerryneedell, 2019-04-04:
No worries from me ... I know how to open an issue if it comes back ;-)

jerryneedell, 2019-04-10:
closing since it appears to be working now.

closed 2019-04-10 by jerryneedell