adafruit/circuitpython #1691: displayio - pause display refresh (for setting up a bunch of items)

ladyada, 2019-03-24:
the opposite of refresh_soon() :)

deshipu, 2019-03-24:
maybe a context manager would work for this, pausing the display for the duration of executing the code inside of it

deshipu, 2019-03-24:
something like:
```python
with display.frame: # pauses the refresh here
   ...
# unpauses and schedules a new refresh here
```

tannewt, 2019-03-26:
I expect to experiment a bunch with frame synchronization in 4.x once we can actually update a frame faster than the display does. Thanks for making an issue for it.

tannewt, 2019-08-02:
Ok, so I think I know what I want to replace `wait_for_frame` and `refresh_soon` with. I think we want an explicit `refresh` call that blocks until new pixels have been sent to the display (but not wait on epaper refresh.)

For fast screens it'll have a `target_fps` kwarg that takes a target frame rate and skips the pixel update if the time since the last frame is more than the gap needed for the frame rate. It'll also have an `max_delay` that will cause an exception if the time since the last updated frame is greater than the given delay.

ladyada, 2019-08-02:
do you want to have anything for pausing so that if you have text on top of a bitmap you dont have a few characters appear at a time

tannewt, 2019-08-02:
@ladyada You don't need to pause because a refresh will only happen when you call `refresh`. It won't auto happen at all. (though the terminal/supervisor will call it too)

ladyada, 2019-08-02:
ohh, wait, so you will have to poke the display for each drawing? how will that work if you draw/change a bitmap like with turtle?

tannewt, 2019-08-02:
You'll want to call refresh from time to time and the dirty area will be updated. I don't think it's too common to have code that may run much faster than the display can update.

ladyada, 2019-08-02:
totally cool - i just wanted to know what to expect - the arduino code's like this too so i'm pretty used to it!

deshipu, 2019-08-03:
That seems reasonable, and much less magical, I like it. I would probably rename ``max_delay`` to ``max_lag``, to make it clearer.

tannewt, 2019-08-03:
I think we will want a `ready_for_refresh` for EInk so we can decide not to wait three minutes.

ladyada, 2019-08-03:
yes, ready_for_refresh is a good idea :) just a note that sharp memory displays *must* be refreshed once a second, or the image disappears. so its sorta the opposite of e-ink.

tannewt, 2019-08-12:
We're going to add `auto_refresh` as a kwarg and property to Display so that `refresh` is optional from the start.

commit 36a23e0f (2019-08-27): Rework refresh API and factor common display stuff out

NOT TESTED! Just compiles

Fixes #1691