adafruit/circuitpython #1696: XBox Adaptive Controller needs custom HID Descriptor & tweaks

ATMakersBill, 2019-03-25:
[Note: @dhalbert requested this issue even though it's really not a bug in CP, but in the XAC]

I spent a great deal of time getting the XBox Adaptive Controller (XAC) to work with CircuitPython.  I found that there were a number of issues with the XAC's implementation of HID that I had to work around.

I have reported those issues to Microsoft, but I've heard back that there will not be a quick resolution to them.  So, in the meantime, if we want to be able to use CP on the XAC (which I do!), it would be great to have my modifications available to other CP users.

Note: **YES**, I agree this is "coding to the bugs" and not a good long-term solution.  But the XAC is making real strides in the AT community, and what we can do with an XAC and $10 of Adafruit parts is amazing: https://www.youtube.com/watch?time_continue=1&v=pJfZb7XVAfM

To make the XAC work with CP, I made the following changes: 

- Modify the HID Descriptor to have ONLY a Gamepad, no Keyboard, Mouse, or Media devices
- Modify the Gamepad classes to expose only a single Joystick (not two) and 8 buttons (not 16)
- Modify the range of the Joystick values.  CP defaults to -127 through 127 and the XAC requires 0 through 65535
- Remove Logical Minimum and Maximum from the HID Descriptor (not sure this was necessary)

I have a build that works that I forked from an early 4.x build.  I have NOT gotten the same code to work on 3.x (although I could try further - once the 4.x works, I dropped it).  I will try to bring that up to the latest 4.x build and share that as a fork.

As for how to proceed, here's an approach that I think will work (not sure it's ideal).

1. Create an XBoxAC class that extends Gamepad and changes the ranges, joystick counts, and buttons.
2. Create an alternative descriptor that does not expose the Keyboard, Mouse, and Consumer components, but instead declares a Gamepad descriptor with the modifications I made above.
3. Create a method to switch between the two descriptor.  Two options here: Build flags are fine if we're OK with sharing UF2 files with every guide.  Alternatively, it should be possible to choose which descriptor to use in boot.py.

Personally, I'd like to see the switch possible at runtime, and I'd like it available on every target board.  I have made XAC controllers from CPXs, Trinkets, ItsyM4s and I would like to make them from an NRF52.  If we have to share custom builds it will be messy.

Again, I recognize this is a hack.  I have pushed on Microsoft on this (and will continue to), but the response has been a clear "don't wait for those changes"..

Thanks
Bill

ATMakersBill, 2019-03-25:
Here are the issues I sent to Microsoft (Bryce Johnson) about how they could improve the HID support in the XAC.  He thanked me and it's in the pipeline, but honestly it's not going to be a speedy fix.  
[XBox Feedback.pdf](https://github.com/adafruit/circuitpython/files/3005031/XBox.Feedback.pdf)

tannewt, 2019-03-26:
For now please make a fork available that bakes in the other descriptor.

I would like to support this through custom USB descriptors in boot.py in the long run. We'll re-prioritize issues after 4.0.0 and hopefully have a better timeline for this then. I'd like to do it sometime this year because it will close #231, our second oldest bug too.

hathach, 2019-05-29:
comment to subscribe since I am doing hid gamepad example in arduino :)

PR #2185 (2019-10-03): Update tinyusb to fix gamepad;add HID OUT interface descriptor
Fixes #1696.

- Needed to update tinyusb to bring in HID fixes necessary for gamepad to work on XAC controller.
- Added HID OUT interface descriptor to match many existing devices. Not yet supported by CircuitPython libraries.

To use the XAC-compatible gamepad descriptor, add a line like this to, say, `ports/atmel-samd/boards/circuitplayground_express/mpconfigboard.mk`:
```
USB_HID_DEVICES=XAC_COMPATIBLE_GAMEPAD
```
closed 2019-10-03 by tannewt