adafruit/circuitpython #1702: ESP8266 messages still present in translations

deshipu, 2019-03-25:
I have noticed that the translation template still contains all the messages for functions related to WiFi configuration on the ESP8266. Since the support for ESP8266 is getting removed in this version, perhaps we should also remove the messages, so that the translators don't have to do unnecessary work, and to save space?

dhalbert, 2019-03-26:
We should remove the esp8266 port from the translation files. However, the strings don't show up at all in anything that's built, because they're port-specific and not included in non-esp8266 ports. So it appears it's a waste of human time, but not flash space.

dhalbert, 2019-03-28:
Fixed by #1716.

closed 2019-03-28 by dhalbert