adafruit/circuitpython #1704: displayio.Group.sort()

deshipu, 2019-03-25:
It would be nice to have it, to be able to, for example, sort the sprites by their ``y`` coordinate.

tannewt, 2019-03-26:
Note that this is `.sort` so it happens in place. `sorted(group)` would create a new object.

@deshipu Does this need to be native? You could make a Group subclass to do it instead.

deshipu, 2019-03-26:
It's a bit difficult to do fast sorting when all you have is access to individual elements...

How about a different approach — instead of having Group as the container, have Group.items that would be a regular Python list — then you get all the list methods for free, including slices and sort.

tannewt, 2019-03-26:
The challenging bit of C sorting is that the key function will likely be python still. (How would you choose what to sort by otherwise?)

I've thought about making Group more closely tied to list post-4.0.0. I don't like static allocation it does now.

deshipu, 2019-03-26:
I could keep a helper list on the side, sorting it and updating the group with it, but currently that would involve removing all elements from the group one-by-one, and then re-adding them one-by-one from the list. Slice assignment could be great for this (``group[:] = sorted(group)``) I suppose.

closed 2021-04-02 by deshipu