adafruit/circuitpython #1705: displayio.Group rotation and flipping

deshipu, 2019-03-25:
It would be nice to be able to rotate the Group in 90° increments, and to be able to mirror and vertically flip it. One way of implementing it would be adding an ``orientation`` attribute, with eight possible values. One bit would control horizontal flip, one bit — vertical, and last bit would control swapping of the ``x`` and ``y`` coordinates (diagonal flip). This way we can have all possible orientations.

tannewt, 2019-03-26:
Would you prefer Group over TileGrid support?

deshipu, 2019-03-26:
I have no preference, both work for me, and I suppose ``TileGrid`` is a little easier to implement, but having it on ``Group`` would be more consistent given that groups already have coordinate offsets on them. Come to think of it, to have it on groups you would need to have it on tiles first, so maybe we can start with that, and add it to the group later if it is needed.

For me the use case is to save on the amount of graphics stored in memory for things that rotate or are animated by mirroring (most walking animations in my games, etc.)

tannewt, 2019-03-26:
Totally. I plan on adding them. It's just a question of where. Starting with TileGrid makes sense to me.

One challenge will be OnDiskBitmap because reads in one direction are waaaaaayyyy faster than the other.

PR #1940 (2019-06-13): Add partial display update support.
Different operations to the display tree have different costs. Be
aware of these costs when optimizing your code.
* Changing tiles indices in a TileGrid will update an area
covering them all.
* Changing a palette will refresh every object that references it.
* Moving a TileGrid will update both where it was and where it moved to.
* Adding something to a Group will refresh each individual area it
covers.
* Removing things from a Group will refresh one area that covers all
previous locations. (Not separate areas like add.)
* Setting a new top level Group will  [... cut]
commit eb21fc3e (2019-06-13): Add partial display update support.

Different operations to the display tree have different costs. Be
aware of these costs when optimizing your code.
* Changing tiles indices in a TileGrid will update an area
covering them all.
* Changing a palette will refresh every object that references it.
* Moving a TileGrid will update both where it was and where it moved to.
* Adding something to a Group will refresh each individual area it
covers.
* Removing things from a Group will refresh one area that covers all
previous locations. (Not separate areas like add.)
* Setting a new top level Group will [... cut]
closed 2019-06-13 by dhalbert