adafruit/circuitpython #1709: Can't connect to REPL with Debian 9

dhalbert, 2019-03-26:
Some users have been unable to connect to the REPL from Debian Linux machines. This problem may be related to #1642 and #1617.

See @mlupo commenting in https://forums.adafruit.com/viewtopic.php?f=60&t=149291#p737722. MTO in discord #help-with-circuitpython 2019-03-25 also had trouble with two different Debian 9 machines, a Udoo Ultra (x64) running Debian 9.7, and an older laptop running 9.something. MTO did not have an issue with an Odroid HC2 running 9.7.

dhalbert, 2019-03-26:
I installed Debian 9.8 on an external hard drive, and then booted it up on three machines:

1. Dell Optiplex 7010 desktop
2. Dell Latittude 4310 laptop
3. Lenovo 110S laptop

In all cases, I was able to connect a PyPortal running 4.0.0 beta.5, and then connect to the REPL via `/dev/ttyACM0`. So I'm not sure what's going on with the other machines. I've asked @MTO on discord to update to 9.8 to see if that fixes things.

Tagging @hathach

emteeoh, 2019-03-26:
On the Odroid, uname -a reports
Linux hc2-2 4.14.87+ #1 SMP PREEMPT Thu Dec 20 14:41:09 UTC 2018 armv7l GNU/Linux

On the Udoo: 
uname output before and after:
Linux OfficeOfMTO 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux
Linux OfficeOfMTO 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux

I don't see a change in behaviour: The filesystem doesn't mount, the serial port doesn't respond. Here's dmesg output:
[ 1077.084324] usb 1-1: new full-speed USB device number 20 using xhci_hcd
[ 1077.301689] usb 1-1: New USB device found, idVendor=239a, idProduct=8032
[ 1077.301707] usb 1-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[ 1077.301720] usb 1-1: Product: PyPortal
[ 1077.301731] usb 1-1: Manufacturer: Adafruit Industries LLC
[ 1077.301740] usb 1-1: SerialNumber: B223FB755364933502020213334480FF
[ 1077.309634] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[ 1077.312582] usb-storage 1-1:1.2: USB Mass Storage device detected
[ 1077.313607] scsi host2: usb-storage 1-1:1.2
[ 1077.335969] input: Adafruit Industries LLC PyPortal as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8032.0008/input/input18
[ 1077.399213] hid-generic 0003:239A:8032.0008: input,hidraw2: USB HID v1.11 Keyboard [Adafruit Industries LLC PyPortal] on usb-0000:00:14.0-1/input3
[ 1077.399717] usb 1-1: USB disconnect, device number 20
[ 1078.351115] usb 1-1: new full-speed USB device number 21 using xhci_hcd
[ 1078.494336] usb 1-1: New USB device found, idVendor=239a, idProduct=8032
[ 1078.494354] usb 1-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[ 1078.494370] usb 1-1: Product: PyPortal
[ 1078.494379] usb 1-1: Manufacturer: Adafruit Industries LLC
[ 1078.494387] usb 1-1: SerialNumber: B223FB755364933502020213334480FF
[ 1078.497369] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
[ 1078.499334] usb-storage 1-1:1.2: USB Mass Storage device detected
[ 1078.500087] scsi host2: usb-storage 1-1:1.2
[ 1088.541724] hid-generic 0003:239A:8032.0009: timeout initializing reports
[ 1088.542683] input: Adafruit Industries LLC PyPortal as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8032.0009/input/input19
[ 1088.604233] hid-generic 0003:239A:8032.0009: input,hidraw2: USB HID v1.11 Keyboard [Adafruit Industries LLC PyPortal] on usb-0000:00:14.0-1/input3
[ 1100.318651] cdc_acm 1-1:1.0: failed to set dtr/rts
[ 1120.800370] cdc_acm 1-1:1.0: failed to set dtr/rts
[ 1120.932780] usb 1-1: reset full-speed USB device number 21 using xhci_hcd

emteeoh, 2019-03-26:
Here's dmesg from the odroid (which works fine)
[4251594.309299] usb 2-1: new full-speed USB device number 2 using exynos-ohci
[4251594.548350] usb 2-1: New USB device found, idVendor=239a, idProduct=8032
[4251594.553786] usb 2-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[4251594.561057] usb 2-1: Product: PyPortal
[4251594.564954] usb 2-1: Manufacturer: Adafruit Industries LLC
[4251594.570593] usb 2-1: SerialNumber: B223FB755364933502020213334480FF
[4251594.584428] usb-storage 2-1:1.2: USB Mass Storage device detected
[4251594.590210] scsi host1: usb-storage 2-1:1.2
[4251594.632118] cdc_acm 2-1:1.0: ttyACM0: USB ACM device
[4251594.636573] usbcore: registered new interface driver cdc_acm
[4251594.641687] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[4251594.651851] input: Adafruit Industries LLC PyPortal as /devices/platform/soc/12120000.usb/usb2/2-1/2-1:1.3/0003:239A:8032.0001/input/input1
[4251594.721822] hid-generic 0003:239A:8032.0001: input,hidraw0: USB HID v1.11 Keyboard [Adafruit Industries LLC PyPortal] on usb-12120000.usb-1/input3
[4251594.733777] usbcore: registered new interface driver usbhid
[4251594.739231] usbcore: registered new interface driver snd-usb-audio
[4251594.745702] usbhid: USB HID core driver
[4251595.616409] scsi host1: scsi scan: INQUIRY result too short (5), using 36
[4251595.621941] scsi 1:0:0:0: Direct-Access     Adafruit PyPortal         1.0  PQ: 0 ANSI: 2
[4251595.631108] sd 1:0:0:0: Attached scsi generic sg1 type 0
[4251595.639406] sd 1:0:0:0: [sdb] 16377 512-byte logical blocks: (8.39 MB/8.00 MiB)
[4251595.650417] sd 1:0:0:0: [sdb] Write Protect is off
[4251595.653965] sd 1:0:0:0: [sdb] Mode Sense: 03 00 00 00
[4251595.658364] sd 1:0:0:0: [sdb] No Caching mode page found
[4251595.662431] sd 1:0:0:0: [sdb] Assuming drive cache: write through
[4251595.745388]  sdb: sdb1
[4251595.787356] sd 1:0:0:0: [sdb] Attached SCSI removable disk

emteeoh, 2019-03-26:
On the laptop: A Lenovo Thinkpad X1 Carbon (4th gen) also running debian 9.8
The kernel was already up to date. Dist-upgrade only updated a few libs.
uname output:
Linux CarbonX1G4 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux

from dmesg when I plug in the pyportal:
[ 4885.145818] usb 1-1: new full-speed USB device number 90 using xhci_hcd
[ 4885.297839] usb 1-1: New USB device found, idVendor=239a, idProduct=8032
[ 4885.297842] usb 1-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[ 4885.297844] usb 1-1: Product: PyPortal
[ 4885.297845] usb 1-1: Manufacturer: Adafruit Industries LLC
[ 4885.297846] usb 1-1: SerialNumber: B223FB755364933502020213334480FF
[ 4885.298604] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 4885.299165] usb-storage 1-1:1.2: USB Mass Storage device detected
[ 4885.299248] scsi host2: usb-storage 1-1:1.2
[ 4885.386376] input: Adafruit Industries LLC PyPortal as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8032.0052/input/input89
[ 4885.447665] hid-generic 0003:239A:8032.0052: input,hidraw0: USB HID v1.11 Keyboard [Adafruit Industries LLC PyPortal] on usb-0000:00:14.0-1/input3
[ 4885.447901] usb 1-1: USB disconnect, device number 90
[ 4886.362363] usb 1-1: new full-speed USB device number 91 using xhci_hcd
[ 4886.504466] usb 1-1: New USB device found, idVendor=239a, idProduct=8032
[ 4886.504476] usb 1-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[ 4886.504483] usb 1-1: Product: PyPortal
[ 4886.504490] usb 1-1: Manufacturer: Adafruit Industries LLC
[ 4886.504495] usb 1-1: SerialNumber: B223FB755364933502020213334480FF
[ 4886.505761] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[ 4886.510939] usb-storage 1-1:1.2: USB Mass Storage device detected
[ 4886.511254] scsi host2: usb-storage 1-1:1.2
[ 4896.674450] hid-generic 0003:239A:8032.0053: timeout initializing reports
[ 4896.675017] input: Adafruit Industries LLC PyPortal as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.3/0003:239A:8032.0053/input/input90
[ 4896.735948] hid-generic 0003:239A:8032.0053: input,hidraw0: USB HID v1.11 Keyboard [Adafruit Industries LLC PyPortal] on usb-0000:00:14.0-1/input3
[ 4908.450516] cdc_acm 1-1:1.0: failed to set dtr/rts

dhalbert, 2019-03-28:
@emteeoh Could you try this build on your pyportal, on the various hosts that don't work?

https://adafruit-circuit-python.s3.amazonaws.com/bin/pyportal/en_US/adafruit-circuitpython-pyportal-en_US-20190328-c6e3b17.uf2. This is the build from this PR: https://github.com/adafruit/circuitpython/pull/1721.

emteeoh, 2019-03-28:
Wahoo! CIRCUITPY mounts, *and* I can talk to it on the console!
It is seeing some garbage on the console somehow, causing errors, but I'm pretty sure that's host side. Once I'm on the console and hit CTRL-D it runs normally.

Could we get a build for the neoTrellis too?

dhalbert, 2019-03-28:
Just look for the same commit number c6e3b17 (or later) on the Amazon S3 site above for other builds.

The garbage you are seeing could be due to `modemmanager` running on your host (especially if you se "AT" in the strings. Just uninstall it -- it doesn't have much purpose nowadays.

closed 2019-03-28 by dhalbert

emteeoh, 2019-03-28:
Thanks!
The neoTrellis works too with c6e3b17!

removing modemmanager fixed the garbage, too.