adafruit/circuitpython #1712: 4.x beta crashes when Trinket M0 plugged into an Android phone OTG port

mgithubreq, 2019-03-27:
Trinket M0 running 4.x beta or HEAD works fine when plugged into a laptop and fails when plugged into a Android phones using an OTG cable.  3.x works fine on the same devices.

dhalbert, 2019-03-27:
@hathach This might be a good lead on some USB issues, and one you could test locally.

dhalbert, 2019-03-28:
@mgithubreq Could you try this build on your Trinket M0?
https://adafruit-circuit-python.s3.amazonaws.com/bin/trinket_m0/en_US/adafruit-circuitpython-trinket_m0-en_US-20190328-c6e3b17.uf2

It fixes a USB HID problem that may also affect connectivity. PR for fix is #1721.

mgithubreq, 2019-03-29:
@dhalbert Thank you!  Unfortunately, I can't test this right now.  We have only one Trinket and I already passed it on to the team doing further integration into our target device...

dhalbert, 2019-03-29:
I tried this on an Amazon Fire, and it no longer seems to crash. Unfortunately, the HID keyboard presented by the Trinket prevents the on-screen keyboard from appearing, and there was no obvious way to allow the on-screen keyboard to appear. I'm going to close this for now, because it sounds similar to the other scenarios for #1721.

closed 2019-03-29 by dhalbert