adafruit/circuitpython #1728: PyPortal Crash into HardFault_Handler

alwayslinux2, 2019-03-30:
Code runs for variable amount of time, then hard crashes, doesn't matter what the except tries to catch. 

One example:
You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython df79e9a on 2019-03-27; Adafruit PyPortal with samd51j20

Other times this:
Traceback (most recent call last):
  File "code.py", line 53, in <module>
ReloadException: 

Where line 53 is just 
      time.sleep(1)

[boot_out.txt](https://github.com/adafruit/circuitpython/files/3025899/boot_out.txt)

[code_py.txt](https://github.com/adafruit/circuitpython/files/3025901/code_py.txt)

dhalbert, 2019-03-30:
Thanks for the report.

1. Could you use beta.6 instead of this intermediate build? Thanks.
2. What is typically a "variable amount of time": seconds, minutes, hours?
3. Could you confirm that your pyportal library is up to date?
4. When running, are you connected to the REPL? If so, does your host computer go to sleep, and then the crash happens?

I'm running your code on a PyPortal, same except I didn't have your background image, so I'm using a different one.

alwayslinux2, 2019-03-30:
I was on adafruit-circuitpython-pyportal-en_US-20190327-df79e9a.uf2
but I also tried adafruit-circuitpython-pyportal-en_US-4.0.0-beta.5
and adafruit-circuitpython-pyportal-en_US-4.0.0-beta.6
The daily seemed to crash within minutes at most.
beta 5 I didn't let run all that long
beta 6 crashed a couple times right away, but then has been running for 30 minutes so far.
[nature.bmp.gz](https://github.com/adafruit/circuitpython/files/3025958/nature.bmp.gz)

My computer hasn't been going to sleep. The dumps I posted were from where I was connected to serial from mu-editor. A couple times It happened after 5 or six minutes being connected to a USB charger with no computer.

dhalbert, 2019-03-31:
I have reproduced some kind of hang after running this for a number of hours. I've restarted it with a debugger to see what I can catch. It was running unattached to USB (power-only cable), and I'm repeating that.

alwayslinux2, 2019-03-31:
I let it run overnight, using beta 6, and it has now run without error for 14 hours. Maybe it was fixed by beta 6.

tannewt, 2019-04-02:
Please reopen if this happens again with Beta 6.

closed 2019-04-02 by tannewt