adafruit/circuitpython #1730: No module named bitbangio

RWeducation, 2019-03-31:
I'm using circuitpython on a custom made arduino zero board. I've got it all working and using the circuitpython version 3.1.2 for arduino zero.

but when I try to import bitbangio, I get this error: 

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'bitbangio'
`

what is the solution to it?

deshipu, 2019-03-31:
Use ``busio`` instead?

RWeducation, 2019-03-31:
> Use `busio` instead?

Does `busio` allow SPI on any pin?

dhalbert, 2019-03-31:
`bitbangio` is omitted from the Arduino Zero build for space reasons. The Zero uses 64kB of internal flash for CRICUITPY, so flash state is very tight, as with Gemma M0 and Trinket M0 It would be possible to add `bitbangio` back and remove something else to make room.

`busio` does not allow SPI on any pin: there are restrictions based on the SPI peripheral on the SAMD21. But if you choose your pins carefully, you can get hardware SPI, and then not need `bitbangio`. See the big table in the front of the SAMD21 datasheet which lists which pins can be used for what.

RWeducation, 2019-04-01:
> `bitbangio` is omitted from the Arduino Zero build for space reasons. The Zero uses 64kB of internal flash for CRICUITPY, so flash state is very tight, as with Gemma M0 and Trinket M0 It would be [...]

makes sense. How would I add this module while compiling it myself?

dhalbert, 2019-04-02:
In ports/atmel-samd/boards/arduino_zero/mpconfigboard.mk, set `CIRCUITPY_BITBANGIO = 1`. If the build then overflows flash, you'll need to turn other things off by setting them to 0. But see if it fits first.

RWeducation, 2019-04-02:
Tried building with `CIRCUITPY_BITBANGIO = 1` and got the following error.

`Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
In file included from ../../shared-bindings/bitbangio/__init__.c:32:0:
../../shared-bindings/bitbangio/__init__.c:88:50: error: 'MP_QSTR_bitbangio' undeclared here (not in a function); did you mean 'MP_QSTR_rotaryio'?
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_bitbangio) },
                                                  ^
../../py/obj.h:168:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 3) | 0x00000006))
                                                        ^~~
../../shared-bindings/bitbangio/__init__.c:88:38: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_bitbangio) },
                                      ^~~~~~~~~~~
../../py/mkrules.mk:55: recipe for target 'build-arduino_zero/shared-bindings/bitbangio/__init__.o' failed
make: *** [build-arduino_zero/shared-bindings/bitbangio/__init__.o] Error 1`

Meanwhile I'll continue exploring what can be done. thanks for the help.

tannewt, 2019-04-23:
@suraj022 Clean and rebuild. The QSTRs don't necessarily get rebuilt when changing flags.

This isn't an issue we can fix on our end.

closed 2019-04-23 by tannewt