adafruit/circuitpython #1738: missing error message `bytes(negative number)`

TG-Techie, 2019-04-02:
I was messing around in python, for fun, and noticed that there was no error message for trying to make a bytes instance of a negative number. instead of saying 'ValueError: cannot convert negative number to bytes' it just raises 'ValueError:'.

this is copied from the repl on mu.
```
>>> bytes(-5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: 
```
running: os: macos, cirpy: beta4 built on the first of april from master.

deshipu, 2019-04-03:
MicroPython (and by inheritance also CircuitPython) has many places where it throws exceptions without error messages, or re-uses more generic error messages than regular Python has. This is a trade-off, necessary to keep the binary small. It seems to me that in this particular case it's relatively easy to realize what the reason for the error is.

tannewt, 2019-04-16:
I agree with @deshipu. No extra error message is needed here for now.

closed 2019-04-16 by tannewt