adafruit/circuitpython #1744: displayio.Bitmap wrapping after certain displayio.Palette size

caternuson, 2019-04-04:
Tested on a PyPortal with:
```
Adafruit CircuitPython 4.0.0-beta.6 on 2019-03-30; Adafruit PyPortal with samd51j20
```
Test code:
```python
import board
import displayio

COLORS = 16

palette = displayio.Palette(COLORS)
palette[0] = 0x000000
palette[1] = 0xFFFFFF

bitmap = displayio.Bitmap(320, 240, COLORS)
bitmap[(100, 100)] = 1
bitmap[(100, 101)] = 1
bitmap[(101, 100)] = 1
bitmap[(101, 101)] = 1

tile_grid = displayio.TileGrid(bitmap, pixel_shader=palette)
group = displayio.Group()
group.append(tile_grid)
board.DISPLAY.show(group)
board.DISPLAY.refresh_soon()
```
run the above get the expected little square:
[image]
change this line to:
```python
COLORS = 17
```
run again and some kind of repetition happens:
[image]

makermelissa, 2019-04-04:
Oh, I thought it was just me.

caternuson, 2019-04-04:
FYI - The mentioning of `displayio.Palette` is a red herring. It's simply triggered by the `value_count` parameter when creating the Bitmap instance. Anything above `16` creates the behavior.

Also, direct indexing vs. using (x,y) tuple does not matter.

closed 2019-04-04 by dhalbert

caternuson, 2019-04-05:
Nice! Thanks for the fix. And now my hacks to `adafruit_imageload` seem to be working better also :)
[image]

PR #1746 (2019-04-04): Support multi-byte values with Bitmap
It also corrects the behavior of single byte values.

Fixes #1744
commit 6fcda1de (2019-04-04): Support multi-byte values with Bitmap

It also corrects the behavior of single byte values.

Fixes #1744