adafruit/circuitpython #1749: Windows 10 descriptor request failed

siddacious, 2019-04-05:
I have a feather m0 express with Beta 6 that when first plugged into my windows 10 machine will show a "USB device not recognized" popup. In the device manager under usb controllers it shows up as "Unknown USB Device (Device Descriptor Request Failed)".

If I single press the reset, it boots as normal with the mass storage drive and serial port.
Once working, if I unplug and replug right after it will show up as expected but if I wait approximately 20 seconds after unplugging before replugging it, it will either not show any activity or show the "not recognized" error.

I can do more testing as needed.

hathach, 2019-04-05:
@siddacious does it happend with metro m0 express as well, I have the metro, could help to troubleshoot this.

dhalbert, 2019-04-05:
@siddacious Could you try other ports on that host and other machines, if possible?

siddacious, 2019-04-05:
@hathach I wasn't able to reproduce it on a metro m0 using the same port and cable.

siddacious, 2019-04-05:
@dhalbert I tried on one of the front ports (previously it was a rear port on my optiplex 7020) and I can get it to happen there as well.

After many plug-unplug cycles it appears that the best way to get it to happen is to slowly insert the micro usb so that the board is powered before the plug is fully seated. I'm guessing this is because it boots before the D+/- has a solid connection.

siddacious, 2019-04-05:
I just tested on my mac and a dramatically slow insertion was enough to get it to not enumerate correctly and show "AppleUSBHostPort::disconnect: persistent enumeration failures" in dmesg. Hitting reset makes it show up as expected.

By "dramatically slow" I mean going slow enough to stop as soon as the charge light comes on, waiting for 5 seconds or so and then continuing to plug in.

dhalbert, 2019-04-05:
Is it worse than you'd expect on the Windows 10 box (i.e., does it happen all the time), or is it just you were plugging in slowly at first? We could add a small delay somewhere before we bring up USB. The M4 is too fast :smiley: .

siddacious, 2019-04-05:
Just to clarify, this is with an **m0**

I only started paying attention because It happened several times during normal use. At first I thought it was always the first time I plugged it in after boot but then discovered more ways of making it happen.

tannewt, 2019-04-18:
@siddacious Would you mind testing the latest master with your Windows box?

siddacious, 2019-04-18:
will do

siddacious, 2019-04-18:
@tannewt still present with current master. Normal speed insertion shows a "USB Device not recognized" error and shows up in the device manager as "Unknown USB Device (Device Descriptor Request Failed)"

dhalbert, 2019-04-18:
If you plug it in fast, it doesn't happen?

siddacious, 2019-04-18:
Well of course now it works/doesn't happen at any speed! I tested a good 5-8 times before making the previous comment but now it's behaving itself.

dhalbert, 2019-04-18:
Can you try it with a different cable, especially one that might be of different manufacture? I had a flaky cable once that disconnected when tilted.

siddacious, 2019-04-18:
Scratch that, it got busted again. Now it's about 50/50 working or not plugging in as fast as I can which is still about "normal"

siddacious, 2019-04-18:
@dhalbert I've tried with two different cables both in good condition from reputable sources and that worked with other boards just fine. They're both 2 meters, is it worth testing with a shorter cable?

dhalbert, 2019-04-18:
No, that's fine. But are you plugging/unplugging on the board end or the host end? If the board end, try the host end, since it's conceivable it's a problem with the board connector in that case.

kevinjwalters, 2019-04-18:
Perhaps too obvious, but have you checked the upgrade to beta 7 was applied ok? E.g. look at REPL output or boot_out?

hathach, 2019-04-19:
@siddacious do you have another  **feather m0 express** to test with, just in case the current board has some defect. I once bend a usb port when plugging cable to hard :D

siddacious, 2019-04-19:
@hathach I don't have another feather m0 currently but I'm going to grab one in my next order

tannewt, 2019-05-08:
@siddacious Mind checking this in the next few days since we're back from PyCon?

siddacious, 2019-05-08:
If I recall correctly my next step in testing was to try and get a good wireshark capture. I'll try to do that this afternoon

dhalbert, 2021-06-15:
Closing this as not seen now.

closed 2021-06-15 by dhalbert