adafruit/circuitpython #1750: Allow displayio to treat all SPI parameters as commands

makermelissa, 2019-04-05:
The SSD1331 likes everything sent as commands. I could format the INIT data as all commands, but that gets lengthy, so it would be nice to just pass a parameter to allow this to happen automatically. Additionally all of the boundary parameters need to be treated as commands too.

deshipu, 2019-04-05:
I think all displays want parameters as commands. Data is strictly only for pixel data.

tannewt, 2019-04-05:
I checked the ST7735R datasheet and it says parameters are data. (See the D/C column below.)

[image]

deshipu, 2019-04-05:
You are right, I tested it with DC=0 and it doesn't work on that display then. Sorry for the assumption.

PR #1751 (2019-04-05): Allow parameter data to be treated as commands for the ssd1331
This fixes #1750
closed 2019-04-05 by dhalbert