adafruit/circuitpython #1752: MicroPython external C modules

dmazzella, 2019-04-05:
why build MicroPython external C modules capability was removed?

interest in a PR for it?

dhalbert, 2019-04-05:
Nothing was removed. Could you point to the MicroPython PR you have in mind? You're talking about an import of a native module, right? We haven't merged from upstream in several months.

dmazzella, 2019-04-05:
[#4195](https://github.com/micropython/micropython/pull/4195/files)

dhalbert, 2019-04-05:
OK, thought you were talking about https://github.com/micropython/micropython/pull/1627 or similar. That PR is after our last merge from micropython. I see this is a build-convenience mechanism: you still need to know how to implement native modules.

Our build structure is slightly different these days, so there would need to be some changes on the original MicroPython PR. If you'd like to work on a PR, that would be fine, maybe with a sample module as well. Do you have an existing specialized module you're trying to compile in?

dmazzella, 2019-04-05:
PR is #1754, for test you can see [USER_C_MODULES](https://docs.micropython.org/en/latest/develop/cmodules.html)

dmazzella, 2019-04-05:
[cmodules.zip](https://github.com/adafruit/circuitpython/files/3049406/cmodules.zip)

unzip into root folder of circuitpython:

```bash
MBP-di-Damiano:~/projects/circuitpython$ make -j12 BOARD=pca10059 SD=s140 USER_C_MODULES=$(pwd)/cmodules -C ports/nrf/
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Including User C Module from /Users/damianomazzella/projects/circuitpython/cmodules/example
```

usage:

```python
Adafruit CircuitPython 4.0.0-beta.6-43-gb9e37818e-dirty on 2019-04-05; PCA10059 nRF52840 Dongle with nRF52840
>>> 
>>> 
>>> import example
>>> example.
__class__       __name__        add_ints
>>> example.add_ints(1, 2)
3
>>>
```

dmazzella, 2019-04-19:
thanks for merge #1754

closed 2019-04-19 by dmazzella