adafruit/circuitpython #1762: displayio interferes with audioio

deshipu, 2019-04-07:
I tried to make a game that would have both music and graphics. I decided to make a simple Tetris game for the Hallowing. The code is here: https://github.com/deshipu/hallowing-tetris

Unfortunately, when the display is being refreshed, that interferes with audioio playback and the result is distorted sound.

deshipu, 2019-04-07:
Here is a recording: https://youtu.be/_ghOrXVjnwM

tannewt, 2019-04-08:
Thanks for testing this! We may need longer audio buffers but partial display updating should help as well.

iraytrace, 2019-04-14:
I can confirm the same problem.  Buffer length may not address the issue.  Consider where a display update starts just prior to playing audio.  For example, on a Hallowing: 

    print('playing sound file')
    wave_file = open(filename, 'rb')
    wave = audioio.WaveFile(wave_file)
    audio.play(wave)

The print() statement will start a display update, and once the audio.play() routine starts it is already too late.  It seems like audio is stuck with bad buffer values before it even starts.  I would suggest the following:

1. API to determine if a display update is in progress so we can avoid actions that would interfere.
2. API to pause the display update and resume it when we need to preempt (like in audioio)
Maybe...
3. Improvements to background task handling, interrupts, setting priorities so background tasks can be more readily overlapped.

Frankly, I'd be thrilled with 1 and 2

kevinjwalters, 2020-07-02:
I have similar issues but I've been shrinking sizes and reducing stepping to make it less apparent.