adafruit/circuitpython #1764: Use micro SD card as main storage ??

bobricius, 2019-04-08:
Hi, I am just interesting if is possible create device without spiflash.
Just MCU+SD card and run code directly from card and also save files to card.

Thanks

dhalbert, 2019-04-08:
Yes, you can import code or read data from an SD card, but, there is no mechanism for accessing the card via the USB port on the device. That is, there's no passthrough from USB to the card filesystem.

bobricius, 2019-04-08:
And is possible modify cpython to access to SD card, without use internal flash and spiflash ?  ... simple device samd21e18+microsd card slot. I am just interesting about this possibility, SD card with fat system is more standardized and have unlimited storage.

dhalbert, 2019-04-08:
The firmware has to live in internal flash, and there needs to be some code to set up using the SD card (though that could be built into the the firmware).

In general, having a large SD card doesn't necessarily help a lot, because the size of programs is limited by the size of RAM, especially on a SAMD21 (32kB). If you have large data files (e.g., data you are streaming out), then the SD card can help. But having 2MB of CIRCUITPY is not that limiting. It's also faster than the SD card, though I couldn't quote you numbers right now.

bobricius, 2019-04-08:
Thanks for fast answer.
I know that firmware need live internal flash, I just asking if is possible replace spiflash with sd card.
If I understand, it is possible but spi flash is better ;)

BriscoeTech, 2019-04-20:
> there is no mechanism for accessing the card via the USB port on the device. That is, there's no passthrough from USB to the card filesystem

Hello dhalbert,

What would need to be done to enable this functionality? It seems like the plumbing is already there since it works with spi flash chips already.

I have some projects that need to be enclosed in project boxes, yet the sd card will be full of sounds and config files that need to be edited from time to time. It would be great to do this over usb without having to open the boxes back up.

dhalbert, 2019-04-20:
CIRCUITPY is a single filesystem volume. The SD card is another storage volume, and so isn't visible inside CIRCUITPY.

It is conceivable we could do this, but we'd have to think about it. One issue is whether the SD card volume would be readonly. To use it safely while the program is running, it would need to be mounted readonly to the CircuitPython code, in case you were editing it over USB at the same time.

We can reopen this as a long-term feature. I'll discuss it with the other core developers.

dhalbert, 2019-10-31:
See https://github.com/adafruit/circuitpython/issues/2206#issuecomment-539725561 for a more thorough explanation of why this is very hard. Closing for now.

closed 2019-10-31 by dhalbert