adafruit/circuitpython #1765: audiobusio.I2SOut erroneous error message

bebeaux, 2019-04-08:


tannewt, 2019-04-08:
The errors are from here: https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/audiobusio/I2SOut.c#L146

They should probably be made less pin specific since it's checking for the combination of the two.

C47D, 2019-04-09:
Should this fix it?
```diff
diff --git a/ports/atmel-samd/common-hal/audiobusio/I2SOut.c b/ports/atmel-samd/common-hal/audiobusio/I2SOut.c
index af84bb807..bbe33c916 100644
--- a/ports/atmel-samd/common-hal/audiobusio/I2SOut.c
+++ b/ports/atmel-samd/common-hal/audiobusio/I2SOut.c
@@ -147,7 +147,7 @@ void common_hal_audiobusio_i2sout_construct(audiobusio_i2sout_obj_t* self,
         mp_raise_ValueError(translate("Invalid bit clock pin"));
     }
     if (ws_clock_unit == 0xff) {
-        mp_raise_ValueError(translate("Invalid bit clock pin"));
+        mp_raise_ValueError(translate("Invalid ws clock pin"));
     }
     if (bc_clock_unit != ws_clock_unit) {
         mp_raise_ValueError(translate("Bit clock and word select must share a clock unit"));
```

tannewt, 2019-04-09:
@C47D probably! We should test it to be sure. I'd change the messages to `"Invalid %q pin", MP_QSTR_bit_clock` style to so that there is only 1 message to translate. (save a little space too)

C47D, 2019-04-09:
Ok, I will fix it and send a PR if you don't mind.

commit 5b94b772 (2019-04-17): Merge pull request #1793 from C47D/fix_audiobusio_msg

Fix I2SOut message, fix #1765
commit 9f631253 (2019-04-17): Fix I2SOut message, fix #1765
closed 2019-04-17 by tannewt