adafruit/circuitpython #1766: Support PWM on TCx_0 pins

ladyada, 2019-04-08:
currently used for the counter TOP, maybe we don't care as much about freq (just within the prescale divider ranges)

tannewt, 2019-04-08:
We currently don't support the first capture compare channel right now because we use it to store the 16bit top value to get precise frequency.

raquo, 2024-03-30:
Hello,

It looks like this issue is the reason why my Feather M4 express can not do PWM on pins A2, A4, and A5 (they're TC4[**0**], TC0[**0**], and TC1[**0**] respectively according to [this chart](https://learn.adafruit.com/assets/111181), so it matches up).

Could you please update the Feather m4 express pinout documentation [here](https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51/pinouts) to reflect this limitation? Currently it says:

> A2 thru A5 - These are each analog input as well as digital I/O pins. These pins can also do PWM.

But, of those, only A3 can do PWM in CircuitPython.

I tested each pin where I expected PWM support with this `code.py` (on CircuitPython 8.2.10):

```python
import board
import pwmio
pwmio.PWMOut(board.A2, duty_cycle=0, frequency=100)
```

A2, A4, and A5 pins were the only ones that threw "ValueError: All timers for this pin are in use" error.

dhalbert, 2024-03-30:
> Could you please update the Feather m4 express pinout documentation [here](https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51/pinouts) to reflect this limitation? Currently it says:

Now noted in the Learn Guide. Thanks.

dhalbert, 2025-01-31:
Closing this because there's a reason it's not supported (to get precise frequency), and it's been documented.