adafruit/circuitpython #1767: NameError occurs and persists after any syntax error

Joe-Davidson1802, 2019-04-08:
I am running a NeoTrellis with CircuitPython 4.0.0 Beta 6.
Whenever I create any kind of compilation error I get the following

Adafruit CircuitPython 4.0.0-beta.6 on 2019-03-30; Adafruit Trellis M4 Express with samd51g19
>>> estingOpen
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'estingOpen' is not defined:

Even if I fix the syntax issue, I still get that error, even if I revert to a previously working code.py file, I have to erase and reinstall circuit python every time.

tannewt, 2019-04-09:
Does the board crash after the traceback? what is the code.py you are using?

tannewt, 2019-04-09:
And thanks for testing Beta 6 @Joe-Davidson1802 ! We'll get these bugs squished soon.

DTV96Calibre, 2019-04-11:
I'm also experiencing this issue ✋

tannewt, 2019-04-12:
@DTV96Calibre What version are you using and what is your code.py? Can you try a build from S3? It'll include a number of fixes. (I plan on releasing it tomorrow.)

DTV96Calibre, 2019-04-12:
@tannewt I was using 4.0.0-beta.6
Unfortunately after reflashing and fixing a bug in my code I haven't been able to reproduce the bug (I've been saving directly to the board without any version control so I can't look back at the old code). I'm dealing with a new bug though where the interpreter doesn't seem to see some attributes of objects.
That new bug isn't addressed in S3

tannewt, 2019-04-12:
Thanks @DTV96Calibre! Let me know if the original issue happens again. I save directly to the disk as well but usually have the file saved in my editor while I'm working on it.

Joe-Davidson1802, 2019-04-12:
@tannewt I will try and recreate in S3, I can reproduce it pretty reliably, so will let you know.

Joe-Davidson1802, 2019-04-12:
> Does the board crash after the traceback? what is the code.py you are using?

I don't believe so, it just does the flashing light thing where it shows you the line number where the issue is. Which is line 1 of "<stdin>" apparently. And it was a few different code.py files as it has happened about 5 times, just as soon as I encountered any kind of syntax exception, like mis-spelling a function name this happens and I have to reflash.

tannewt, 2019-04-15:
@Joe-Davidson1802 Did you manage to reproduce from the latest S3 build or Beta 7? Sounds like the filesystem is getting wiped or at least code.py is.

Joe-Davidson1802, 2019-04-15:
@tannewt Sorry, no chance over the weekend should be able to give it a try tonight.

Joe-Davidson1802, 2019-04-15:
@tannewt Awesome seems to be working now. I won't be having to be so careful about making syntax mistakes now!

closed 2019-04-15 by Joe-Davidson1802

tannewt, 2019-04-15:
Great! Glad it's fixed. Thank you for retesting.