adafruit/circuitpython #1769: HID requires connection to REPL to type.

kdb424, 2019-04-09:
I have a keyboard that's built on circuitpython, and while on beta 4, I am able to plug the keyboard in, the code boots, and I can type. When testing on beta 5, 6, or even latest at the time of writing, I have to connect to the REPL for me to see output. Based on the lights on the keyboard, it appears to be running code, but it runs normal after connecting to REPL, so it's hard to check where the issue is.

ItsyBitsyM4
Code used to test was this https://github.com/KMKfw/kmk_firmware

dhalbert, 2019-04-11:
Hi - I tried this with the simple test below as `main.py`, using a PyPortal (so I could see any errors), and I did not see this issue. If I plug the board in, it starts typing `a`'s in whatever area the mouse is in, even without connecting via the REPL. I am using the latest version of master.

Could you make a simple test case, based on your code in some way? Thanks.

kdb424, 2019-04-11:
I'll see if I can sort it out. It may be a phantom issue that actually stems from not getting a value to set that detects if it is connected to USB or just power. If that detection fails, the code may hang there. I'll take a look.

kdb424, 2019-04-11:
@dhalbert I think that I sorted out what the issue was.
https://github.com/KMKfw/kmk_firmware/blob/8bb55c84cdb00bbe8ffc6e4ba00d133df5a100c1/kmk/firmware.py#L142

return supervisor.runtime.serial_connected
I think I was exploiting a bug and using it as a feature in beta 4. This was used to see if the board was powered directly by 5V power or had USB data lines available. In beta 4, this returns True even if I am not connected to a REPL, and False if powered my 5V power directly. In the newer betas, I have to actually connect to the REPL to have that return True.

Assuming that this is a correct behavior in the current betas, what is the recommended way to check if it's powered by 5V, or has a full USB connection variable, even if not connected to by REPL?

dhalbert, 2019-04-12:
@kdb424 We use a low-level call to the TinyUSB library, `tud_hid_generic_ready()` to see if there's an HID connection. That value is not exposed directly, but it's used by the HID report sending code, which will throw an exception if it's not available after waiting two seconds. See https://github.com/adafruit/circuitpython/blob/master/shared-module/usb_hid/Device.c#L48.

So you could test this indirectly just by sending a do-nothing report (like a keyboard report with no keys pressed), and include a try-catch to catch the exception.

We could expose a direct status property (akin to `.serial_available`, but for HID), but the indirect method above might be good enough.

dhalbert, 2019-04-12:
Closing this as a bug. If you'd like a new feature, feel free to open the request as a new issue.

closed 2019-04-12 by dhalbert

dhalbert, 2019-04-15:
@kdb424 did you try the "sending a do-nothing report" described above as a workaround for now?

klardotsh, 2019-05-12:
@dhalbert @kdb424 I was able to confirm catching the `OSError` is a valid workaround by testing the following script (ignore the wildly inaccurate "frequency" function, but it gets the job done) against a Feather M4 Express (1) plugged into computer - received no blinks, and (2) plugged into power-only AC adapter - received frantic blinks.

```python
import board
import digitalio
import time
import usb_hid

class HIDUsage:
    KEYBOARD = 0x06
    MOUSE = 0x02
    CONSUMER = 0x01
    SYSCONTROL = 0x80

def flicker(io, freq=15, duration=3):
    total_flickers = freq * duration
    onoff_time = 1000 / freq / 1000
    io.value = False

    for _ in range(total_flickers):
        io.value = not io.value
        time.sleep(onoff_time)

if __name__ == '__main__':
    pin = board.D13
    dio = digitalio.DigitalInOut(pin)
    dio.switch_to_output()

    keyboard_device = None

    for dev in usb_hid.devices:
        if dev.usage == HIDUsage.KEYBOARD:
            keyboard_device = dev
            break

    if not keyboard_device:
        flicker(dio, freq=8, duration=5)

    try:
        keyboard_device.send_report(bytearray(8))
    except OSError:
        flicker(dio, freq=30, duration=20)
    except Exception:
        flicker(dio, freq=2, duration=10)
```