adafruit/circuitpython #1770: UART doesn't block until transmit end.

tannewt, 2019-04-09:
UART.write will return before the last byte is actually transmitted. This is usually fine because it has time to finish while other Python code runs. However, if the write is at the end of the code.py, the SERCOM will be reset early and corrupt the last byte.

We should replace ASF4 here and manage the dma ourselves.

KurtE, 2022-05-15:
Note: I personally would prefer that the Serial writes would queue the data and work asynchronous.  And in those case where you need to need to know that the data has completed before doing something that you would
add the  uart.flush() type capability.  

I have not tried any of this on the SAMD, but for example I just tried out the changes for RPI2040 that allowed the RS485 pin to be logically set.  And to show why it was needed on the RP2040,  here is a sketch running that sets a pin for this and have it change state directly versus, me doing IO pin writes on both sides of the uart write operations.

```python
import time
import board
import busio
import supervisor
import sys
from digitalio import DigitalInOut, Direction, Pull

initial_baud = 115200

if hasattr(board, 'NEOPIXEL'): 
    import neopixel

    led_colors = [
        (255, 0, 0),
        (0, 255, 0),
        (0, 0, 255),
        (255, 255, 0),
        (255, 0, 255),
        (0, 255, 255),
        (255, 255, 255),
        (0, 0, 0),
    ]
    color_index = 0
    led = neopixel.NeoPixel(board.NEOPIXEL, 1)
    led.brightness = 0.3

else:
    led = DigitalInOut(board.LED)
    led.direction = Direction.OUTPUT
    led_colors = None

if hasattr(board, 'TX2'): 
    dir1_pin = board.D2
    dir2_pin = board.D3
    tx2 = board.TX2
    rx2 = board.RX2
elif sys.platform == 'RP2040':
    dir1_pin = board.D4
    dir2_pin = board.D5
    tx2 = board.D6
    rx2 = board.D9
else:
    #ESP32-S3
    dir1_pin = board.IO1
    dir2_pin = board.IO16
    tx2 = board.IO17
    rx2 = board.IO18

uart1 = board.UART()
uart2 = None

#def setup_uarts(baud):
if True:
    baud = initial_baud
    print("*** initialize the uarts baud: ", baud, " ***")
    # we have not yet tried to iniatialize the uarts
    if (uart1 != None):
        uart1.deinit()
    uart1 = busio.UART(board.TX, board.RX, rs485_dir=dir1_pin, baudrate=baud)
    print("After Uart1")

    # on second one roll our own direction pin
    if (uart2 != None):
        uart2.deinit()
    uart2 = busio.UART(tx2, rx2, baudrate=baud)
    print("After Uart2")

#setup_uarts(initial_baud)

uart2_dir = DigitalInOut(dir2_pin)
uart2_dir.direction = Direction.OUTPUT
uart2_dir.value = 0
times_1 = 0
times_2 = 0
loop_count = 0
print("Start loop")

while True:
    start_time = time.monotonic_ns()
    uart1.write(b"abcdefghijklmnopqrstuvwxyz")
    after1 = time.monotonic_ns()
    times_1 += int(after1 - start_time)
    uart2_dir.value = 1
    uart2.write(b"abcdefghijklmnopqrstuvwxyz")
    uart2_dir.value = 0
    after2 = time.monotonic_ns()
    times_2 += int(after2 - after1)
    loop_count += 1
    if (loop_count == 10):
        print(int(times_1/10), " ", int(times_2/10))
        times_1 = 0
        times_2 = 0
        loop_count = 0
        if led_colors != None:
            led[0] = led_colors[color_index]
            color_index += 1
            if color_index >= len(led_colors):
                color_index = 0
        else:
            if led.value:
                led.value = False
            else:    
                led.value = True
    time.sleep(0.025)
```

And here is a Logic analyzer output,  top 3 lines are for UART1 and bottom 3 lines are for UART2
[image]

Which you can clearly see, it would not be easy to make it work manually 

I have not run the same sketch or the like on an SAMD 

Although I now do have a Feather M4 express.

kevinjwalters, 2026-09-29:
I agree with @KurtE , keeping the current, long-standing behaviour makes sense to me for `write()` and a `flush()` would be useful. There could also be an optional arg on `write()` to block until sent.