adafruit/circuitpython #1771: feather_nrf52840 - slideshow not working on tft featherwing

jerryneedell, 2019-04-09:
I  have been testing a "slideshow" using a feather_m4_express and an feather_nrf52840 wiht a TFT featherewing (ILI9341).

It runs fine on the feather_m4_express
but it does not work on the feather_nrf52840
It starts up OK, but only displays a small part of each image then eventually (after a few images) goes to a white screen.

This used to work on the nrf52840, but if I recall correctly, some changes were made to accommodate the BLE timing. 
The REPL response is very sluggish on the nrf52840.   

Both boards show the same output:
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.6-56-g5015036c0 on 2019-04-08; Adafruit Feather M4 Express with samd51j19
>>> 
>>> 
>>> 
>>> import slideshow_sd
cat_rug_320x240.bmp  Size: 153.7 KB
cats_320x240.bmp     Size: 163.3 KB
sibs_320x240.bmp     Size: 153.7 KB
skye320x240.bmp      Size: 153.7 KB
```
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.6-56-g5015036c0 on 2019-04-08; Adafruit Feather nRF52840 Express with nRF52840
>>> import slideshow_sd
cat_rug_320x240.bmp  Size: 153.7 KB
cats_320x240.bmp     Size: 163.3 KB
sibs_320x240.bmp     Size: 153.7 KB
skye320x240.bmp      Size: 153.7 KB

```
here is the code being run on both boards:
``` 
import os
import board
import busio
import digitalio
import displayio
import storage
import adafruit_sdcard
import adafruit_ili9341
from adafruit_slideshow import PlayBackOrder, SlideShow

""" If you want to use an SD card to store the images: """
#spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.D5)
sdcard = adafruit_sdcard.SDCard(board.SPI(), cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
IMAGE_DIRECTORY = "/sd/images"

# We must release any active displays before we use them.
displayio.release_displays()

def print_directory(path, tabs=0):
    for file in os.listdir(path):
        stats = os.stat(path + "/" + file)
        filesize = stats[6]
        isdir = stats[0] & 0x4000

        if filesize < 1000:
            sizestr = str(filesize) + " by"
        elif filesize < 1000000:
            sizestr = "%0.1f KB" % (filesize / 1000)
        else:
            sizestr = "%0.1f MB" % (filesize / 1000000)

        prettyprintname = ""
        for _ in range(tabs):
            prettyprintname += "   "
        prettyprintname += file
        if isdir:
            prettyprintname += "/"
        print('{0:<20} Size: {1:>6}'.format(prettyprintname, sizestr))

        # recursively print directory contents
        if isdir:
            print_directory(path + "/" + file, tabs + 1)

print_directory(IMAGE_DIRECTORY)

# Create the slideshow object that plays through once alphabetically.
display_bus = displayio.FourWire(board.SPI(), command=board.D10, chip_select=board.D9)
display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)
slideshow = SlideShow(display, None,
            folder=IMAGE_DIRECTORY, loop=True,
            order=PlayBackOrder.ALPHABETICAL, dwell=5)

while slideshow.update():
    pass

```

deshipu, 2019-04-09:
I think it may be related to #1758 — the SPI speed is no longer hardcoded, and the nrf's defaults may be wrong for that display? Can you try adding ``display_bus.configure(baudrate=12000000)`` in there before you create the FourWire bus?

jerryneedell, 2019-04-09:
I don't understand -- display_bus is created by the line
```
display_bus = displayio.FourWire(board.SPI(), command=board.D10, chip_select=board.D9)
```
when should I execute the display_bus.configure(baudrate=12000000) - after that?

jerryneedell, 2019-04-10:
@deshipu 
how do I set the baudrate as you suggested?

jerryneedell, 2019-04-10:
nevermind -- this is now working with the current master.
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.6-72-gde38ce798 on 2019-04-10; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> 
>>> import slideshow_sd
cat_rug_320x240.bmp  Size: 153.7 KB
cats_320x240.bmp     Size: 163.3 KB
sibs_320x240.bmp     Size: 153.7 KB
skye320x240.bmp      Size: 153.7 KB
```

closed 2019-04-10 by jerryneedell

dhalbert, 2019-04-10:
I think that #1773 may have fixed this.

deshipu, 2019-04-11:
@jerryneedell sorry, I didn't see your questions. You have it resolved, but just for reference, you would do:

```python
spi = board.SPI
spi.configure(baudrate=12000000)
display_bus = displayio.FourWire(spi, command=board.D10, chip_select=board.D9)
```