adafruit/circuitpython #1792: Request: HID Ready check

kdb424, 2019-04-13:
It would be useful to have a hid ready check similar to how `.serial_avaliable` is exposed. I was exploiting a bug in that function using it as a ready check in a beta, using it as a feature, and was informed in #1769 that this was actually a bug not a feature. KMK (Keyboard software for circuitpython) is using this to detect a master slave architecture in split keyboards to see if it is the uart master or slave simply checking if it has HID, and is therefor the one plugged in, now powered with no USB data lanes. I think this could be useful in more master slave cases as well as HID is always available when the USB is plugged in to a computer.

jpconstantineau, 2021-09-16:
This feature would also be useful for detecting if a nrf52 keyboard should enable BLE or USB HID.

dhalbert, 2023-10-28:
I think `supervisor.runtime.usb_connected` serves this purpose now.

closed 2023-10-28 by dhalbert