adafruit/circuitpython #1801: Black horizontal line on top of TileGrid bitmaps

rdagger, 2019-04-16:
I'm loading a 50x50 pixel color bitmap using displayio.OnDiskBitmap().
[test_bitmap.zip](https://github.com/adafruit/circuitpython/files/3082557/test_bitmap.zip)
A black horizontal line is appearing at the top of the image which should be all white.
[image]
Here's my code:
```
import board
import displayio
from st7735r import ST7735R
import time

spi = board.SPI()
cs = board.D11
dc = board.D10
rst = board.D9

displayio.release_displays()
display_bus = displayio.FourWire(spi, command=dc, chip_select=cs, reset=rst)
display = ST7735R(display_bus, width=128, height=160)
splash = displayio.Group()
display.show(splash)

color_bitmap = displayio.Bitmap(128, 160, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0x00FFFF
background = displayio.TileGrid(color_bitmap, x=0, y=0,
                               pixel_shader=color_palette)
splash.append(background)

with open("/test.bmp", "rb") as f:
    bmp = displayio.OnDiskBitmap(f)
    icon = displayio.TileGrid(bmp,
                                    pixel_shader=displayio.ColorConverter(),
                                    x=39, y=55,
                                    width=1, height=1,
                                    tile_width=50, tile_height=50)
    splash.append(icon)
    time.sleep(10)
```
I've reproduced the problem with other bitmaps.  Hardware is Feather nRF52840 Express.

tannewt, 2019-04-17:
Thank you for the bug report! I've reproduced it and am hunting it now.

tannewt, 2019-04-17:
Classic off by one error when subtracting y from height. PR incoming.

PR #1810 (2019-04-18): Fix off by one error in OnDiskBitmap
It resulted in the first row of pixels being pulled for out of bounds.

Fixes #1801
commit c3725673 (2019-04-18): Fix off by one error in OnDiskBitmap

It resulted in the first row of pixels being pulled for out of bounds.

Fixes #1801
closed 2019-04-18 by dhalbert