adafruit/circuitpython #1806: i2c.try_lock() in REPL causes hard crash when code.py is run

siddacious, 2019-04-17:
I was testing some i2c address code and needed to scan the bus on the REPL, then tried to re-start the code and was sent into safe mode.

code.py:
```
import time
import board
import busio
import adafruit_veml7700

i2c = board.I2C()
veml7700 = adafruit_veml7700.VEML7700(i2c)

while True:
    print("Ambient light:", veml7700.light)
    time.sleep(0.1)

```

To reproduce, have the above `code.py` saved on the device, start a REPL and do the following:

```
import board
i2c = board.I2C()
i2c.try_lock()
i2c.scan() # not needed to repro
```
then CTRL-D to re-start the `code.py`

I after some fussing I was able to get a backtrace:
```
Breakpoint 2 at 0x3143a: file ../../supervisor/shared/safe_mode.c, line 81.
(gdb) c
Continuing.

Breakpoint 1, reset_into_safe_mode (
    reason=HARD_CRASH)
    at ../../supervisor/shared/safe_mode.c:81
81	    if (current_safe_mode > BROWNOUT && reason > BROWNOUT) {
(gdb) bt
#0  reset_into_safe_mode (reason=HARD_CRASH)
    at ../../supervisor/shared/safe_mode.c:81
#1  0x00032822 in HardFault_Handler ()
    at supervisor/port.c:290
#2  <signal handler called>
#3  0x00052b28 in mp_load_method_maybe (
    obj=0x20029a80, attr=859, dest=0x20003574)
    at ../../py/runtime.c:1074
#4  0x000530d2 in mp_load_method (base=0x20029a80,
    attr=859, dest=0x20003574)
    at ../../py/runtime.c:1093
#5  0x00038446 in mp_execute_bytecode (
    code_state=0x20003560,
    inject_exc=<optimized out>) at ../../py/vm.c:368
#6  0x00046902 in fun_bc_call.lto_priv.66 (
    self_in=0x20028c40, n_args=3, n_kw=0,
    args=0x20002d64) at ../../py/objfun.c:284
#7  0x0004e954 in mp_call_function_n_kw (
    fun_in=0x20028c40, n_args=3, n_kw=0,
    args=0x20002d64) at ../../py/runtime.c:624
#8  0x0004ea12 in mp_call_method_n_kw (n_args=2,
    n_kw=0, args=0x20002d60)
    at ../../py/runtime.c:639
#9  0x0004033e in mp_obj_instance_make_new (
    self=0x20029350, n_args=2, args=0x2002fc24,
    kw_args=0x2002fb2c) at ../../py/objtype.c:366
#10 0x00040756 in type_call.lto_priv.9 (
    self_in=0x20029350, n_args=2, n_kw=0,
    args=0x2002fc24) at ../../py/objtype.c:1026
#11 0x0004e954 in mp_call_function_n_kw (
    fun_in=0x20029350, n_args=2, n_kw=0,
    args=0x2002fc24) at ../../py/runtime.c:624
#12 0x0004ea12 in mp_call_method_n_kw (n_args=2,
    n_kw=0, args=0x2002fc1c)
    at ../../py/runtime.c:639
#13 0x000381f4 in mp_execute_bytecode (
---Type <return> to continue, or q <return> to quit---
    code_state=0x2002fc08,
    inject_exc=<optimized out>) at ../../py/vm.c:1015
#14 0x00046902 in fun_bc_call.lto_priv.66 (
    self_in=0x20027840, n_args=2, n_kw=0,
    args=0x20002d44) at ../../py/objfun.c:284
#15 0x0004e954 in mp_call_function_n_kw (
    fun_in=0x20027840, n_args=2, n_kw=0,
    args=0x20002d44) at ../../py/runtime.c:624
#16 0x0004ea12 in mp_call_method_n_kw (n_args=1,
    n_kw=0, args=0x20002d40)
    at ../../py/runtime.c:639
#17 0x0004033e in mp_obj_instance_make_new (
    self=0x20027b00, n_args=1, args=0x2002fe2c,
    kw_args=0x2002fd34) at ../../py/objtype.c:366
#18 0x00040756 in type_call.lto_priv.9 (
    self_in=0x20027b00, n_args=1, n_kw=0,
    args=0x2002fe2c) at ../../py/objtype.c:1026
#19 0x0004e954 in mp_call_function_n_kw (
    fun_in=0x20027b00, n_args=1, n_kw=0,
    args=0x2002fe2c) at ../../py/runtime.c:624
#20 0x0004ea12 in mp_call_method_n_kw (n_args=1,
    n_kw=0, args=0x2002fe24)
    at ../../py/runtime.c:639
#21 0x000381f4 in mp_execute_bytecode (
    code_state=0x2002fe10,
    inject_exc=<optimized out>) at ../../py/vm.c:1015
#22 0x00046902 in fun_bc_call.lto_priv.66 (
    self_in=0x20029a00, n_args=0, n_kw=0, args=0x0)
    at ../../py/objfun.c:284
#23 0x0004e954 in mp_call_function_n_kw (
    fun_in=0x20029a00, n_args=0, n_kw=0, args=0x0)
    at ../../py/runtime.c:624
#24 0x0004ea50 in mp_call_function_0 (fun=0x20029a00)
    at ../../py/runtime.c:598
#25 0x000208fc in parse_compile_execute (
    source=0x709d0, input_kind=MP_PARSE_FILE_INPUT,
---Type <return> to continue, or q <return> to quit---
    exec_flags=32, result=0x2002ffb4)
    at ../../lib/utils/pyexec.c:114
#26 0x00020a72 in pyexec_file (
    filename=0x709d0 "code.py", result=0x2002ffb4)
    at ../../lib/utils/pyexec.c:527
#27 0x0003434e in maybe_run_list (
    filenames=0x200001c4 <supported_filenames>,
    exec_result=0x2002ffb4) at ../../main.c:175
#28 0x000345f0 in run_code_py (
    safe_mode=NO_SAFE_MODE) at ../../main.c:215
#29 0x0003486e in main () at ../../main.c:447
```

siddacious, 2019-04-17:
FWIW the particular driver using the `i2c` object doesn't matter, this was just the example I had handy.

siddacious, 2019-04-17:
I updated the title to reflect further testing; if the `code.py` uses 
```
i2c = busio.I2C(board.SCL, board.SDA)
```
instead of `i2c = board.I2C()`, there will be no issue. Additionally, the `try_lock` in the REPL isn't needed, just the `i2c = board.I2C()`

tannewt, 2019-04-17:
What board is this on? Does it crash if you don't have a driver at all?

tannewt, 2019-04-17:
Ok, I think this is because we weren't resetting the shared busses after the repl is run. This could also be why folks have more trouble if they test from the repl rather than code.py directly.

siddacious, 2019-04-17:
I initially saw it on my itsybitsy m4 but then reproduced it on a metro m4 to get the backtrace

PR #1809 (2019-04-18): Unify VM cleanup code
Before we were skipping reset of the standard busses after boot and
repl which could lead to a crash in a subsequent VM run.

Fixes #1806
commit 92ba5c26 (2019-04-18): Unify VM cleanup code

Before we were skipping reset of the standard busses after boot and
repl which could lead to a crash in a subsequent VM run.

Fixes #1806
closed 2019-04-18 by dhalbert