adafruit/circuitpython #1813: readthedocs build is failing

dhalbert, 2019-04-18:
The readthedocs builds have been failing for some time, and we didn't really notice.
https://readthedocs.org/projects/circuitpython/builds/

The latest failure is:
`ModuleNotFoundError: No module named 'sphinxcontrib.rsvgconverter'`
but if you go back a bit, there's an error about a png file that's not really PNG (?).
```
(/usr/share/texlive/texmf-dist/tex/latex/amsfonts/umsb.fd) [1{/var/lib/texmf/fo
nts/map/pdftex/updmap/pdftex.map}] [2] [1] [2] [1] [2]libpng error: Not a PNG file

Chapter 1.

!pdfTeX error: pdflatex (file ./907b5cd1d6faf78dae6eaa9257a2cc7c50629e51.png): 
libpng: internal error
 ==> Fatal error occurred, no output PDF file produced!
```

tannewt, 2019-04-18:
I've been working on fixing this. By "back a bit" I believe you mean an older build. It was happening because the png was hosted on github and loading it from github returned an html page.

dhalbert, 2019-04-18:
It worked! I just refreshed the rtd page! 

[image]

dhalbert, 2019-04-18:
Fixed by #1811 and #1805.

closed 2019-04-18 by dhalbert