adafruit/circuitpython #1814: CircuitPython 4beta7 hard crash (safemode) when using stage module

wawachief, 2019-04-18:
I experience a hard crash when I use stage module with circuitPython 4 beta 7 recompiled with CIRCUITPY_STAGE = 1 option.
I tried the demo example (ball) and the jumper game. Each time, the game launches and crashes after few minutes.
The crash occurs on a itsybitsy M4 and a metro M4 as well. The display used is adafruit 1.8" Color TFT LCD display with MicroSD Card Breakout - ST7735R (product 358).

deshipu, 2019-04-18:
Can you please attach the exact code you are running?

wawachief, 2019-04-18:
The zip file contains all the flash memory and the firmware used for itsybitsy M4 as well

[bug.zip](https://github.com/adafruit/circuitpython/files/3093959/bug.zip)

deshipu, 2019-04-18:
I could reproduce this with your uf2, but not with an uf2 I compiled myself from master with the stage.py and ugame.py frozen. I will try again with master but without the frozen files.

deshipu, 2019-04-18:
Crashes with master too when the library files are not frozen.

wawachief, 2019-04-18:
I experienced also a freeze with stage library frozen.

deshipu, 2019-04-18:
Can you change your grid definition to:
```python
background = stage.Grid(bank,10,8)
```
And try again? It seems it only happens with a large grid.

wawachief, 2019-04-18:
Same thing. I experienced a crash with grid(10,8) and ugame and stage frozen

deshipu, 2019-04-18:
Mine still has yet to crash, but maybe I'm just lucky — it does seem to be random. I will try to figure out how to connect a debugger to the itsybitsy.

deshipu, 2019-04-18:
Hmm, actually, it didn't crash into safe mode, but the serial comms are frozen...

wawachief, 2019-04-18:
Same here. I am not in safe mode anymore since I froze the stage and ugame modules

deshipu, 2019-04-18:
This is weird, all I get in GDB is:

```
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x0002f700 in run_background_tasks.part ()
(gdb) bt
#0  0x0002f700 in run_background_tasks.part ()
#1  0x00005116 in main ()
```

dhalbert, 2019-04-18:
Can you compile with DEBUG=1 so you get line numbers and symbols?

deshipu, 2019-04-18:
Yeah, doing it now.

deshipu, 2019-04-18:
After a few tries, I managed to get this traceback:
```
Breakpoint 1, HardFault_Handler () at supervisor/port.c:289
289	{
(gdb) bt
#0  HardFault_Handler () at supervisor/port.c:289
#1  <signal handler called>
#2  0x20006f90 in ?? ()
#3  0x0001561a in common_hal_busio_spi_write.part.1 (self=<optimized out>, 
    data=<optimized out>, len=<optimized out>) at common-hal/busio/SPI.c:320
#4  0x2000133c in mp_state_ctx ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

tannewt, 2019-04-18:
Ok, I think I figured it out. The SPI object was getting cleaned up because the `never_free` didn't correctly mark it's first block during a collect. PR coming soon.

PR #1816 (2019-04-19): Correct collect of permanent pointers.
The previous implementation managed to keep all but the head
pointer of the list.

Fixes #1814
commit 54ef87c6 (2019-04-19): Correct collect of permanent pointers.

The previous implementation managed to keep all but the head
pointer of the list.

Fixes #1814
closed 2019-04-19 by dhalbert