adafruit/circuitpython #1819: https://github.com/adafruit/circuitpython/issues

ThomasRKirkland, 2019-04-22:
Crash into the HardFault_Handler.
I was trying to run the Thermometer.py example code on a PyPortal.
Below is the url address of a .png snapshot of the CircuityPython directory

[image]

dhalbert, 2019-04-22:
What version of CircuitPython were you running, and what version of the libraries and Thermometer.py? Thanks.

ThomasRKirkland, 2019-04-22:
The PyPortal_Smart_Thermometer was downloaded yesterday at 10pm from your
web site.
The CircuitPython I was running
adafruit-circuitpython-pyportal-en_US-4.0.0-beta.6.uf2
The library bundle  I was running is
adafruit-circuitpython-bundle-4.x-mpy-20190420

I loaded the Viewmaster example and it is working after unplugging the
pyPortal and then plugging it back it.   I have had other problems with the
Smart Thermometer code to.

dhalbert, 2019-04-22:
Could you try "Absolute Newest" from here: https://circuitpython.org/board/pyportal/ ?

We have fixed several crashes since beta.6. Beta.7 is better, and "Absolute Newest" is even better. We'll be doing another release soon.

ThomasRKirkland, 2019-04-22:
The message that I get now is
Running in safe mode! Not running saved code.
Code done running. Waiting for reload.
soft reboot

The Viewmaster code is also not running correctly.  It displays the first
bmp file and then stops.

I think I will give up for the night.   Thank you for your help.

ThomasRKirkland, 2019-04-22:
I tried the Adafruit Quotes Example program and it works correctly.
I will try again tomorrow.
Thanks again.

ThomasRKirkland, 2019-04-22:
The viewmaster example code is now working correctly.  I had changed the
code to switch pictures based on the light sensor and the room was too dark
when I retried the code the first time.  When I turned on the lights the
code worked correctly.

dhalbert, 2019-04-22:
Closing since it appears newer versions work fine.

closed 2019-04-22 by dhalbert