adafruit/circuitpython #1825: import error : no module named displayio

stephenvowell, 2019-04-24:
I get this error attempting to use the ili9341_simpletest.py I am using aMetro Mo Express and a 2.8 touch shield

ImportError: no module named 'displayio'

https://github.com/adafruit/Adafruit_CircuitPython_ILI9341/blob/master/examples/ili9341_simpletest.py

makermelissa, 2019-04-24:
Hi, which version of CircuitPython are you running this with? `displayio` wasn't added to the metro M0 until CircuitPython 4.0 Beta 4. I would recommend trying the latest CircuitPython 4.0 RC 1

stephenvowell, 2019-04-24:
OK, will give it a try!

stephenvowell, 2019-04-24:
Latest version 4.0 RC 1 same error

stephenvowell, 2019-04-24:
You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

makermelissa, 2019-04-24:
I'm trying it out on a Metro M0 with the latest version as well and attempting to reproduce your error. Did you get the library files from the bundle?

stephenvowell, 2019-04-24:
above error CircuitPython 4.0.0-rc.1

stephenvowell, 2019-04-24:
Yes

stephenvowell, 2019-04-24:
[image]
s.githubusercon

makermelissa, 2019-04-24:
I'm getting `Cannot proxy the given URL` when I click on that link.

stephenvowell, 2019-04-24:
I rebooted no I get the following , we are getting closer

Traceback (most recent call last):
  File "main.py", line 8, in <module>
ImportError: no module named 'adafruit_ili9341'

makermelissa, 2019-04-24:
Ok, the adafruit_ili9341.mpy file should be inside a lib folder inside your CIRCUITPY folder. That's the one from the bundle.

makermelissa, 2019-04-24:
I would also try grabbing the ili9341_simpletest.py file from the examples folder, copy it right into your CIRCUITPY folder and rename it to code.py.

stephenvowell, 2019-04-24:
Yea its not in the the bundle. and I tried I would also try grabbing the ili9341_simpletest.py file from the examples folder, copy it right into your CIRCUITPY folder and rename it to code.py. too

makermelissa, 2019-04-24:
You'll need the latest bundle from https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases since it was just added earlier today.

makermelissa, 2019-04-24:
You can also get it from https://github.com/adafruit/Adafruit_CircuitPython_ILI9341/releases

stephenvowell, 2019-04-24:
new problems

Traceback (most recent call last):
  File "code.py", line 6, in <module>
ReloadException: 
soft reboot

Line 6 is the import board

stephenvowell, 2019-04-24:
All import errors with board and displaio are gone now .Code failed at line 38
Traceback (most recent call last):
  File "code.py", line 38, in <module>
ReloadException: 
soft reboot
Below is the code from the example....
""
This test will initialize the display using displayio
and draw a solid red background
"""

import board
import displayio
import adafruit_ili9341

spi = board.SPI()
tft_cs = board.D9
tft_dc = board.D10

displayio.release_displays()
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs)

display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)

# Make the display context
splash = displayio.Group(max_size=10)
display.show(splash)

color_bitmap = displayio.Bitmap(320, 240, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0xFF0000

try:
    bg_sprite = displayio.TileGrid(color_bitmap,
                                   pixel_shader=color_palette,
                                   position=(0, 0))
except TypeError:
    bg_sprite = displayio.TileGrid(color_bitmap,
                                   pixel_shader=color_palette,
                                   x=0, y=0)
splash.append(bg_sprite)

while True:
    pass

tannewt, 2019-04-24:
ReloadException is caused by writing to the file system and is expected. Once it reloads your code should run without error.

stephenvowell, 2019-04-24:
Yup, The demo is not working on the screen have rebooted many times no luck testing the screen

stephenvowell, 2019-04-24:
Could the display be bad, ?

deshipu, 2019-04-24:
Can you post a photo of your setup? It could be something with the connections.

stephenvowell, 2019-04-24:
Its connected to Ada product 1651 , 2.8 TFT shield... connected to a Metro mo express# 3505, no wires necessary !

boot_out.txt
code.py
CURRENT.uf2
examples
lib
README.txt
adafruit_bus_device
adafruit_hid
adafruit_ili9341.py
adafruit_dotstar.mpy
[Archive.zip](https://github.com/adafruit/circuitpython/files/3111987/Archive.zip)

Here you go

stephenvowell, 2019-04-24:
The shield and TFT screen arrived today, The demo app that came with the Metro MO Express worked fine. attached the TFT shield and downloaded the adafruit_ili9341.py to test and that's when the code started not seeing the board and the other modules.

Installed CircuitPython 4.0.0-rc.1 code saw the seeing the modules and uploaded to board without errors, but no demo on the screen.

reloaded the working demo that was shipped with the Mo Express it stopped working, seems to be two Issues Both with the boot Phyon versions

deshipu, 2019-04-24:
Looks like you have the DC and CS pins swapped compared to how that shield is connected. DC should be ``board.D9`` and CS should be ``board.D10``.

stephenvowell, 2019-04-24:
I didn't change the code, just copied from GitHub and dropped in the drive. I will chang the code and see if that works

deshipu, 2019-04-24:
The code wasn't written for that particular shield, you have to adapt it.

stephenvowell, 2019-04-24:
Back to the same error 
    Traceback (most recent call last):
    File "code.py", line 7, in <module>
    ImportError: no module named 'displayio'

changed the code to
  spi = board.SPI()
 tft_cs = board.D10
 tft_dc = board.D9

I think my board hates me!

deshipu, 2019-04-24:
Did you reflash the older version of the firmware in the mean time? What does the ``boot_out.txt`` file say?

stephenvowell, 2019-04-24:
Adafruit CircuitPython 2.2.0 on 2018-01-02; Adafruit Metro M0 Express with samd21g18

deshipu, 2019-04-24:
That is an old version. Please flash the 4.0.0RC1 from https://circuitpython.org/board/metro_m0_express/ — to do that, download the UF2 file from that website, press reset on your board twice quickly, wait for the METROBOOT disk to appear, and copy thet UF2 file to it.

stephenvowell, 2019-04-24:
The same result here is the boot out txt

  Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit Metro M0 Express with samd21g18

stephenvowell, 2019-04-24:
Traceback (most recent call last):
  File "code.py", line 7, in <module>
ImportError: no module named 'displayio'

deshipu, 2019-04-24:
I don't have a metro m0 express to test it myself, but on all other boards the ``displayio`` module is present.
What does it say what you enter ``help("modules")`` into the REPL console?

stephenvowell, 2019-04-24:
Can't enter the repl with mu. for some reason. the serial monitor section of mu isn't responding

stephenvowell, 2019-04-24:
I will send you my metro m0 express !

stephenvowell, 2019-04-24:
Got the RELP working

>>> help ("modules")
__main__          builtins          i2cslave          rtc
_os               busio             math              storage
_pixelbuf         collections       microcontroller   struct
_time             digitalio         micropython       supervisor
analogio          displayio         neopixel_write    sys
array             errno             os                terminalio
audiobusio        fontio            pulseio           time
audioio           frequencyio       random            touchio
bitbangio         gamepad           re                usb_hid
board             gc                rotaryio          usb_midi
Plus any modules on the filesystem
>>>

stephenvowell, 2019-04-24:
I recreated the error

   You are running in safe mode which means something unanticipated happened.
   Looks like our core CircuitPython code crashed hard. Whoops!
   Please file an issue at https://github.com/adafruit/circuitpython/issues
  with the contents of your CIRCUITPY drive and this message:
  Crash into the HardFault_Handler.

[drive contents.zip](https://github.com/adafruit/circuitpython/files/3112258/drive.contents.zip)

stephenvowell, 2019-04-24:
>>> help("modules")
__main__          builtins          i2cslave          rtc
_os               busio             math              storage
_pixelbuf         collections       microcontroller   struct
_time             digitalio         micropython       supervisor
analogio          displayio         neopixel_write    sys
array             errno             os                terminalio
audiobusio        fontio            pulseio           time
audioio           frequencyio       random            touchio
bitbangio         gamepad           re                usb_hid
board             gc                rotaryio          usb_midi
Plus any modules on the filesystem
>>>

deshipu, 2019-04-24:
Looks like the ``displayio`` module is there. If Mu couldn't connect, maybe you were seeing an old error still from the 2.0.0 version.

stephenvowell, 2019-04-24:
just reinstalled every thing, no errors, but no demo on TFT as well

rebooted my mac and reinstalled Mu
boot out text is 

Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit Metro M0 Express with samd21g18

stephenvowell, 2019-04-24:
I tested the display on a uno and display works...the demo does not work on Metro Mo express

deshipu, 2019-04-24:
I'm out of ideas, sorry, maybe someone else will be able to help.

stephenvowell, 2019-04-24:
Thanks, I was concerned that there was something wrong with the display...but I tested the display on a  uno and the display works fine. Metro Mo express works as well looks like it is just the demo/hardware combo that's not

deshipu, 2019-04-24:
I suspect something more may require adjusting in that particular example code.

stephenvowell, 2019-04-24:
Should I close the issue or leave it

deshipu, 2019-04-24:
We can always close it later. By the way, you may also get some help on the Adafruit forum.

stephenvowell, 2019-04-24:
ok

hexthat, 2019-04-24:
Is this a RAM issue? Metro M0 non express has less RAM right?

dhalbert, 2019-04-24:
@stephenvowell That shield is meant for classic Arduino. In the product description it says:

>Works with any classic Arduino '328. Solder closed three jumpers to use the ICSP header for use with Leonardo or Mega

See https://learn.adafruit.com/adafruit-2-8-tft-touch-shield-v2/connecting#tft-screen-pins-2-4
and
https://learn.adafruit.com/adafruit-2-8-tft-touch-shield-v2/connecting#using-with-a-mega-slash-leonardo-2-12

`board.SPI()` would use the 2x3 header pins. You could change the code to create an SPI device on the classic Arduino SPI pins, or change the jumpers

stephenvowell, 2019-04-24:
Thanks, But I am using Metro Mo express and CircuitPython problem is not with the shield but with the sample code, Plus I am using the release can 4.x python its a little funny..... and Mu Ide doesn't like it..

deshipu, 2019-04-24:
It's actually a problem with the code not fitting the shield (or vice versa), so you have two options here: either you change the jumpers on the shield, as the article says, or use an SPI object created on the pins that the shield uses by default, instead of using the board's default SPI pins. The first one requires a change in hardware, the second one in code — it's up to you which one you choose.

stephenvowell, 2019-04-24:
Lets try with code,

deshipu, 2019-04-24:
You have to change this line:

```python
spi = board.SPI()
```

to this:

```python
import busio

spi = busio.SPI(clock=board.D13, MOSI=board.D11)
while not spi.try_lock():
    pass
spi.configure(baudrate=24000000)
spi.unlock()
```

dhalbert, 2019-04-24:
Here's the change (not tested), using the pin numbers mentioned in https://learn.adafruit.com/adafruit-2-8-tft-touch-shield-v2/connecting#tft-screen-pins-2-4

```python
...
import busio

spi = busio.SPI(board.D13, MOSI=board.D11, MISO=board.D12)  # SCK, MOSI, MISO
...
```

stephenvowell, 2019-04-24:
Thanks Guys, I will let you know how it goes, of to work

stephenvowell, 2019-04-24:
Well I did get a little color change, with this change "import busio
spi = busio.SPI(board.D13, MOSI=board.D11, MISO=board.D12)  # SCK, MOSI, MISO"

Nothing with this
"import busio

spi = busio.SPI(clock=board.D13, MOSI=board.D11)
while not spi.try_lock():
    pass
spi.configure(baudrate=24000000)
spi.unlock()'

same result ...

My have to go with JUMPERS

stephenvowell, 2019-04-24:
But before I do  here is the code Mu doesent see errors.

import board
import displayio
import adafruit_ili9341
import busio

spi = busio.SPI(board.D13, MOSI=board.D11, MISO=Board.D12) #SCK, MOSI, MISO

tft_cs = board.D9
tft_dc = board.D10

displayio.release_displays()
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs)

display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)

# Make the display context
splash = displayio.Group(max_size=10)
display.show(splash)

color_bitmap = displayio.Bitmap(320, 240, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0xFF0000

try:
    bg_sprite = displayio.TileGrid(color_bitmap,
                                   pixel_shader=color_palette,
                                   position=(0, 0))
except TypeError:
    bg_sprite = displayio.TileGrid(color_bitmap,
                                   pixel_shader=color_palette,
                                   x=0, y=0)
splash.append(bg_sprite)

while True:
    pass

stephenvowell, 2019-04-24:
Bye the way, Thank you so much for helping me figure this out

dhalbert, 2019-04-24:
@stephenvowell put your code inside triple backquotes, and it will format nicely. I will edit your previous postings to do that.

```
    ```
    Python code here
    ```

stephenvowell, 2019-04-26:
Thanks DHALBERT...
I cut the jumpers and added the others  all works now

[1 more reply]

closed 2019-04-26 by dhalbert