adafruit/circuitpython #1826: Crash into the HardFault_Handler.

stephenvowell, 2019-04-24:
Using CircuitPython 4.0.0-rc.1

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

deshipu, 2019-04-24:
Thank you for the crash report. Can you attach the contents of your CIRCUITPY drive too, please?

stephenvowell, 2019-04-24:
[Archive.zip](https://github.com/adafruit/circuitpython/files/3112245/Archive.zip)
I recreated the error

tannewt, 2019-05-08:
@stephenvowell what board was it running on?

tannewt, 2019-05-08:
Ah, I see the boot_out.txt says Metro M0. I'm unable to reproduce it. Please reply if it still happens on the latest S3 build and I'll reopen.

closed 2019-05-08 by tannewt