adafruit/circuitpython #1828: displayio: support for monochrome displays

deshipu, 2019-04-24:
Since I'm building a board with an OLED screen, and I already have a working prototype, I thought that I would start a discussions about the possible ways of implementing support for monochrome (1-bit) displays in ``displayio``.

I'm using an SH1106 OLED display, but it is organized very similar to SSD1306 OLED, the PCD8544 "Nokia" LCD, and the e-ink displays, so we should be able to cover a good range.

The current display code, especially the ``displayio_refresh_displays`` function, is written with a lot of assumptions about how the display is organized, and that doesn't match the monochrome displays.

The largest difference is in that we no longer can update individual pixels — we have to update a whole byte, eight pixels, at a time. That requires us to choose one of the following approaches:

- Only allow updates that are aligned to byte boundaries.
- Allow updates that are not aligned to byte boundaries, but automatically align them.
- Only allow full screen updates.

Realistically speaking, with 16 times less data, and with much smaller displays than the color ones, the last option would be more than fast enough and the simplest to implement.

A second difference is in how we choose where the data is written. The color displays have a command that lets us select an rectangular area of the screen that is going to be filled with the data we send. The monochrome displays work differently: they have a cursor that we can position within the memory with dedicated commands, and the data we send will be written to that place. Depending on the mode, the cursor may automatically advance and may automatically wrap around to the next page, but not all displays support all the modes. In addition, the commands are usually encoded in a single byte with the desired position, so for example to select the page for the cursor, we have eight different commands, each selecting a different page.

Lastly, the video memory is usually organized differently than the bitmap buffer in ``displayio`` assumes. Instead of pixels being arranged in bytes horizontally, like this:

```
76543210 76543210 76543210 76543210

76543210 76543210 76543210 76543210

76543210 76543210 76543210 76543210
```

They are arranged into pages, like this:

```
0 0 0 0 0 0 0
1 1 1 1 1 1 1
2 2 2 2 2 2 2
3 3 3 3 3 3 3
4 4 4 4 4 4 4
5 5 5 5 5 5 5
6 6 6 6 6 6 6
7 7 7 7 7 7 7

0 0 0 0 0 0 0
1 1 1 1 1 1 1
2 2 2 2 2 2 2
3 3 3 3 3 3 3
4 4 4 4 4 4 4
5 5 5 5 5 5 5
6 6 6 6 6 6 6
7 7 7 7 7 7 7
```

The simplest implementation would have the ``displayio_refresh_displays`` create a buffer for a single page of the display, fill it with pixels based on the groups, send the commands for setting the cursor to the right page and column 0, and send that page. Repeat for all the pages.

Of course practically whole logic contained in that function would need to be completely different for that — probably we would have a separate subclass of ``Display``, and have all the logic moved into there, with the refresh just calling the functions from that class?

tannewt, 2019-04-25:
How do eink's fit into this picture? They are slow to refresh and some allow partial updates.

`refresh_displays` will likely change a lot with partial screen update support. My current thinking is to structure the partial areas into rectangles rather than individual pictures. By passing rectangles areas can fill in multiple pixels in a single call. I think they'll also want a pixel orientation to dictate the pixel order in the buffer. (OnDiskBitmap may want to will it in non-linearly if pixels are off the disk in a different orientation.)

deshipu, 2019-04-25:
I have limited experience with e-ink screens, having only used the smaller ones, but at least those are very similar to the other monochrome displays. Here is a driver I wrote for the gew0154t8 chip: https://gist.github.com/deshipu/f98123276336c6763d66c439761ccd48

Notable differences:
- You need to provide both old pixel data and new pixel data for them to be able to calculate the optimal update pattern. That means we will need double buffering.
- They have a "busy" pin, which indicates that the display is still refreshing and not accepting commands.
- They have an additional command to actually trigger updating the display. You can control what kind of update it is going to be (fast/slow, partial/full) by setting some registers just before that. In particular, you need to set the lookup table of voltages for different pixel transitions.
- Two-color displays basically have two monochrome images in their memory, at different addresses, one for black/white, and one for red/white or yellow/white.

makermelissa, 2019-04-25:
I’m a little late to the party here, but could another option be to read the 8-bit block of pixels, update whatever bits and rewrite that modified byte? It’s a similar approach that I took to the RTC FeatherWing so that each element of the current time could be modified individually.

deshipu, 2019-04-25:
Many of those displays don't have the MISO pin in the SPI mode, and even if the display has it, the breakouts they are mounted on don't make it available. Also, it would be painfully slow.

For a 128x64 SSD1306 OLED, you need 1kB of RAM to keep a full copy of the whole screen, or 128B to keep a buffer for a single line. Compared to a 160x128 ST7735's 40kB of RAM, that is more reasonable.
Besides, with 1kB of data to compute and send to the display each frame, doing full-screen updates doesn't sound as bad. And for full-line updates that is only 128B, so even more reasonable.

deshipu, 2019-04-25:
Right now the most expensive part of refreshing the screen is actually computing the colors of all the pixels, and not actual sending of the computed data (which could be, if it became a problem, done asynchronously with DMA anyways). If we keep that 1kB buffer of the whole screen for the monochrome displays, we can only update the pixels that changed, and keep the old pixels from previous frames. We would still send full bytes to the display, but that's fast. If we don't keep that buffer and instead align the updated areas to full bytes, we will need to also compute the parts of those bytes that didn't change.

deshipu, 2019-04-25:
Note that this math works differently for the 4-bit displays, which require only 4 times less memory per pixel than the 16-bit ones, instead of 16 times less like the 1-bit displays do, so we might need to rethink that for them.

Tasm-Devil, 2019-05-01:
Hi,
I'd also like to use displayio with an ssd1306.

makermelissa, 2019-05-01:
Yep, that’ll probably be one of the first displays since it’s so ubiquitous.

commit 6797ec6e (2019-07-23): Add support for grayscale displays that are < 8 bit depth.

This also improves Palette so it stores the original RGB888 colors.

Lastly, it adds I2CDisplay as a display bus to talk over I2C. Particularly
useful for the SSD1306.

Fixes #1828. Fixes #1956
closed 2019-07-23 by dhalbert