adafruit/circuitpython #1830: 4.0.0 rc1 goes into safe mode after a normal installation

kevinjwalters, 2019-04-25:
I took a CPX running 4.0.0 beta7 connected to Windows host:

1. reset it to put 4.0.0 rc 1 on it
2. connected to serial console
3. deleted lib
4. copied lib from adafruit-circuitpython-bundle-4.x-mpy-20190425 onto board
5. ran chkdsk f:  (it said no problems)
6. renamed code.py on the board to something.py
7. copied myfile.py onto board and renamed it to code.py
8. hit control-D

This popped out:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are running in safe mode which means something unanticipated happened.
The reset button was pressed while booting CircuitPython. Press again to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit CircuitPlayground Express with samd21g18
>>> import os
>>> os.listdir()
['.fseventsd', '.metadata_never_index', '.Trashes', 'boot_out.txt', 'System Volume Information', 'code-error-demos-buffer-bug.py', 'code-PREMESSAGESPLIT.py', 'waveforms-PREMESSAGESPLIT.mpy', 'synthfunctions-PREMESSAGESPLIT.mpy', 'synthfunctions.mpy', 'code-PUTBACK.py', 'code-one-of-the-examples.py', 'code-prechannelproperty.py', 'code-sometimes-blows-up-memory-issues.py', 'K2', 'code-pre-hijack-for-synth.py', 'code.py']
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
Running in safe mode! Not running saved code.

You are running in safe mode which means something unanticipated happened.
The reset button was pressed while booting CircuitPython. Press again to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit CircuitPlayground Express with samd21g18
```

Hit C-D again and I seem to be stuck in a cycle here and can't escape safe mode.

**boot_out** still says

```
Adafruit CircuitPython 4.0.0-beta.7 on 2019-04-13; Adafruit CircuitPlayground Express with samd21g18
```

kevinjwalters, 2019-04-25:
Actually it's telling me (I think) to press the reset button to exit safe mode? I'm still puzzled as to why boot_out would not say 4.0.0 rc1?

I'm going to unmount F: and press reset button

kevinjwalters, 2019-04-25:
boot_out now says:

```
Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit CircuitPlayground Express with samd21g18
```

and all appears fine but it's not the smoothest of upgrades. Is what happened to me expected?

dhalbert, 2019-04-25:
Did you do `storage.erase_filesystem()` previously? I'd recommend that over just `chkdsk`. When you copy lots of files (like the `lib/` dir), you need to do an Eject on Windows and wait for the "it's safe to remove" message. Do that whenever you copy a file, as opposed to editing a file with a safe editor.

kevinjwalters, 2019-04-25:
This is a different CPX board.

kevinjwalters, 2019-04-26:
As an experiment I've gone back to 4.0.0 beta 7 and then re-upgraded to rc1 and that all went smoothly.

For some reason windows mounts this as CIRCUITPY **F:** but CPLAYBOOT _**E:**_ - that's a little confusing.

Another small curiosity. I have also noticed that the first time I _read_ boot_out.txt it triggers an Auto-reoad. Both **notepad** and DOS command **type** do this.

tannewt, 2019-04-26:
Cool, thanks for retesting. Will close this issue.

The read triggering a reload is likely an OS thing. It may be updating the last access time of the file for example.

closed 2019-04-26 by tannewt

kevinjwalters, 2019-04-26:
Does a FAT file system have an _access_ time?

dhalbert, 2019-04-26:
It shouldn't be enabled by default. See footnote 2 here: https://en.wikipedia.org/wiki/File_Allocation_Table.

However, I see the reload on `type d:boot_out.txt` as well, so something is going on. We should see if we can distinguish these touches from a regular write. It may be touching the dirty bit on the filesystem, though I'm not sure why.

kevinjwalters, 2019-04-26:
It didn't seem to do it on a _second_ or _third_ repeat of the command.  That suggests _day resolution_ access dates could be a factor but inspecting file system would show this one way or t'other. I didn't even know these were present, they probably added those after I stopped playing with FAT12/FAT16 rubbish.

kevinjwalters, 2019-05-19:
@dhalbert Did you create a separate bug/enhancement issue ticket to look at the file read triggering a reload? I've just noticed even inspecting properties from the Windows Explorer GUI triggers a reload.

kevinjwalters, 2019-05-19:
The file system of my PyPortal just got corrupted by me renaming a file as reported by **chkdsk**. I renamed `cache.bmp` to `cache-iffy.bmp`. That triggered a reload, of course. It's also more complicated because this is the nasa image viewer from Adafruit Learn and that makes the storage accessible and writeable. I've tried a `chkdsk e: /f` whilst at REPL prompt and it claims to be better now.

dhalbert, 2019-05-20:
> @dhalbert Did you create a separate bug/enhancement issue ticket to look at the file read triggering a reload? I've just noticed even inspecting file properties (was looking at `/cache.bmp` on a PyPortal) from the Windows Explorer GUI triggers a reload.

There is metadata that's written by Windows sometimes (like access times) that we can't distinguish from other writes. If you have some file monitoring, indexing, or anti-virus software on your machine, it can aggravate this. I don't think this is a new issue, but I'll investigate further.

> The file system of my PyPortal just got corrupted by me renaming a file as reported by **chkdsk**. I renamed `cache.bmp` to `cache-iffy.bmp`. That triggered a reload, of course. It's also more [...]

>  It's also more complicated because this is the nasa image viewer from Adafruit Learn and that makes the storage accessible and writeable.

Yes, basically this is dangerous, and especially so on windows. Don't write files while the NASA image viewer is running.

kevinjwalters, 2019-05-20:
It's odd though that a properties view triggers that.

I've been trying to debug a problem with PyPortal / Adafruit IO, see https://forums.adafruit.com/viewtopic.php?f=60&t=151923. I need to be able to preserve images from the file system when it misbehaves for debug.

Years ago a colleague proposed mounting a ufs filesystem read-only from a second host concurrently. I had to explain why even that wasn't safe as there were no read locks, no memory cache coherence and worse case we could cause very unexpected things in the worst case leading to kernel panics. I'd imagine we are in the same boat here with the humble FAT12?